Blueforcer/awtrix3 #298: [FEATURE REQUEST] add an option to mirror display

volkerrichert, 2023-09-05:
# Feature Request

I have my awtrix sitting on a shelf above my bed. During night the time can't be read easily. It would be fine to have an option to make it more "readable". This will also help to mount it upside down. 

### Describe the solution / feature you'd like

Add an option to mirror the display at the x- and/or y- axis. This can be two options, one for each axis or just one with 4 possible settings (none / mirror X / mirror Y / mirror both)

Blueforcer, 2023-09-05:
There is already a mirror and flip function
https://blueforcer.github.io/awtrix-light/#/dev

closed 2023-09-05 by Blueforcer

volkerrichert, 2023-09-05:
Ahh, i missed that. THX

volkerrichert, 2023-09-05:
ah, ok. Just at boot time. Is there a change to change ist via API at runtime?

Blueforcer, 2023-09-05:
Unfortunately not

volkerrichert, 2023-09-05:
sound promising..