Blueforcer/awtrix3 #301: [FEATURE REQUEST] Customizable automatic brightness control

socke7, 2023-09-07:
# Feature Request

### Is your feature request related to a problem?

I built a DIY version of the clock and realize the screen brightness is way too low for my hardware if set to auto mode. Even if I directly  illuminate the LDR with a torch, I'm getting a brightness of only 44/255 ("bri") while "ldr_raw" is 1023 which seems to be the maximum.
I'm aware of the min_brightness/max_brightness properties, but these seem to work in a different way - please correct me, if I'm wrong.

### Describe the solution / feature you'd like

Please, give us the option to make the screen brighter in auto mode. Probably the min_brightness/max_brightness properties could be rededicated in a way that these are the brightness es at min / max sensor value.

### Describe alternatives you've considered

Alternatively there could be a curve (stored as list) of multiple points  describing the dependency of sensor value and LED brightness.

### Additional context

None.

robercy, 2023-09-07:
Hi. Maybe you can change the divider resistor to a better value that solve your problem.

socke7, 2023-09-07:
Yes, I already thought on this. But as stated before even if the sensor signal is at maximum, LED brightness is just 44/255. From what I see, this can't be adjusted in hardware.

robercy, 2023-09-07:
And the LDR resistance difference between darkness and full bright?

socke7, 2023-09-07:
It's between a few 100k and a bit below 100Ohm. At average brightness around 10k - as it should be for a 5516 type.

Blueforcer, 2023-09-13:
min_brightness/max_brightness sets the range for the autobrightness wich can be 0-255

closed 2023-09-13 by Blueforcer