Blueforcer/awtrix3 #303: [FEATURE REQUEST] Please unveil correct paramters to flash FW via esptool.py

Grothesk242, 2023-09-07:
I tried to flash my clock using this command but the flash failed: 

```
esptool --port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 --chip esp32 --no-stub --before default_reset --after hard_reset write_flash --flash_mode dout --flash_freq 40m --flash_size detect 0x0 ulanzi_TC001_0.84.bin
```
Can you please tell what parameters need to be adjusted? Thx!

Blueforcer, 2023-09-07:
`esptool.py" --chip esp32 --port "COM7" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 ulanzi_TC001_0.84.bin`

Its easier to use the webflasher.

Grothesk242, 2023-09-07:
Thx! Will try it. 
The webflasher is not working for me, Datatransfer via USB is not permitted by my companies security policy. So I have to switch to a headless machine that I have access to.

zzarne, 2023-09-07:
May be unrelated, but I could't use the webflasher when the original firmware was installed. Device was detected fine but it always timed out before writing anything. esptool from Ubuntu didn't work either. pip(3) install esptool and then using esptool.py did something - I didn't know the correct way so i softbricked it. But it was erased and from then on, the webflasher worked fine. If I'm not the only one affected by this, maybe this should go in a FAQ.

Grothesk242, 2023-09-07:
Just one more thing: 
Where to get the bootloader.bin partitions.bin boot_app0.bin files? :-)

Blueforcer, 2023-09-07:
https://github.com/Blueforcer/awtrix-light/tree/main/docs/ulanzi_flasher/firmware

Grothesk242, 2023-09-07:
Thx! Worked like a charm. I only had to add --no-stub option because my Debian is missing the stubs.

closed 2023-09-07 by Blueforcer