Blueforcer/awtrix3 #304: [BUG]  animated icon -> blinking text

matze20211, 2023-09-08:
# Bug report

## Describe the bug

custom app with animated icon and text (used since version 0.63), the text is blinking like the animation, without icon, text stays

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.85 

## To Reproduce

Steps to reproduce the behavior: remove icon, and activate icon

## Expected behavior

The text should not blinking

matze20211, 2023-09-08:
tested with icon 12144.gif and 44321.gif:
`
set myBroker publish awtrix_d66fe0/custom/1 {"text":"[MQTT2_DVES_6CA8FD:BME280_Temperature] °C","icon":"12144","color":"#FFFF00"})`

Blueforcer, 2023-09-08:
Known bug. Hot fix is on its way

Blueforcer, 2023-09-08:
https://github.com/Blueforcer/awtrix-light/releases/tag/0.84