Blueforcer/awtrix3 #309: [FEATURE REQUEST] MultiColorText as in AWTRIX

vosch86, 2023-09-10:
# Feature Request

### Is your feature request related to a problem?

When I checked the API for AWTRIX I didnt notice it was the wrong one at first, it wasnt for the light version. I tried the MultiColorText option and - of course - it didnt work. So meanwhile I got to this conclusion but wonder if I am the only person looking for that feature.
Background: My goal is to show the surrounding temperature and humidity in one message with indicating colors. When I did it in single steps with each own durations, I set the color of the text to a related meaning, e.g. humidity < 40% = too dry, color red/orange, between 40% and 60% good, color green, humidity > 60% too high, color blue.  
I would like to combine this now with the temperature, showing e.g. "22°C / 65%" and have the "22°C" part green for comfortable temp and "65%" part blue for too moist. 

### Describe the solution / feature you'd like

I would like to have the option from AWTRIX to send a data package with different texts and colors for the idea above. Feed is currently given by home assistant via MQTT.

### Describe alternatives you've considered

It is already possible to send all that one after another, but when you have too many values/messages, it takes too much time (for me) to wait for the last one to come again. So I started combining them but as visible in the screenshot, I can only set one color. I am still new to this so if anyone has another idea or way or if miss something in the documentation, please let me know.
Thanks a lot for this project and looking forward to your feedback.

### Additional context

Current setup, scrolling is a bit sticky here, but idea should be clear.
[image]

Blueforcer, 2023-09-10:
Check text fragments in API docs.

closed 2023-09-10 by Blueforcer

vosch86, 2023-09-10:
My bad, didn't check for fragments, only searched for "multi" and "color". Sorry for bothering