Blueforcer/awtrix3 #310: [BUG]

DrRSatzteil, 2023-09-11:
# Bug report
Unexpected WiFi behaviour after lost connection

## Describe the bug

WiFi is Switches off in parts of our house during night time so the Ulanzi clock is without connection for a couple of hours. In the morning it shows "AP MODE" on the screen. So far this is the expected behaviour I guess.

The thing is: the device is actually not in AP mode. I see that it has an active dhcp lease and I can even access the web frontend where it also shows me that it is connected to my wifi network. All other features of the device are non-functional (API, screen mirror, etc.) until after a manual reboot.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)

## To Reproduce

Steps to reproduce the behavior:
Turn off WiFi until Clock enters AP mode. Then reactivate WiFi. Clock is reconnecting successfully but thinks it's still in AP mode.

## Expected behavior

Preferably leave AP mode and commence normal operation.

## Screenshots

N/A

## Logs

```
I don't have any logs so far. If required I'm happy to create a log file. 
```

## Additional context

N/A

ofenboecka, 2023-09-11:
Thanks DrRSatzteil, I have a similar setup with WiFi turned off at night.
It's true that after re-enabling WiFi, awtrix reconnects and I'm able to access the web interface - and initiate a reboot there.
But it's not responsive to http API calls, a reboot fails with "FileNotFound":

$ curl -X POST -H 'Content-Type:application/json' http://[awtrixIP]/api/reboot --data '{ }'
FileNotFound

Therefore, the reboot cannot be automated.
Is there a way to completely turn off this automatic AP mode?

Thanks, br

Xeno22, 2023-09-14:
Same here - since two days my Awtrix greets me in the morning with "AP MODE" enabled.
No idea if this happens because of the new Awtrix light 0.86 or the new UniFi firmware ...
It seems to disconnect and reconnect, but Awtrix stays in AP MODE:
[image]

closed 2023-09-14 by Blueforcer

Blueforcer, 2025-07-03:
Awtrix only enters AP mode, if it can't connect to the wifi **at boot **. There is no further check during runtime.
So either your awtrix runs out of power or there is a different problem.

kevstone122, 2025-07-03:
Okay this i also recognize if sometimes the clock doesn't get that fast an IP from DHCP, then it changes to the AP Mode. 
But today my mesh network got hang and so there was no internet anymore available. All clocks have "AP Mode" on the Display. 
I reboot the mesh system and then i have to reboot each clock. It looks like, that the clock is changing during runtime into the AP Mode. They are always on the USB C Port.

[image]

commit 3a9c8cb9 (2023-09-14): V0.87

### ✨ Features
- 🎵 Adds `/rtttl` API to play a rtttl melody string directly
- 🖥️ Adds HTTP fullscreen liveview at `/fullscreen` with optional fps parameter

### 🐛 Bug Fixes
- 🎨 Fixes Matrix color for HA entity. Closes #314
- 🎨 Fixes standard indicator colors
- 🌈 Fixes moodlight bug. Closes #311
- 📆 Fixes drawing coordinates for time calendar icon. Closes #312
- 🔡 Fixes a bug with the free pixel between icon and text
- 🟠 Fixes circle drawing
- 📝 Fixes a bug while using textOffset and pushIcon. Closes #305

### 🔥 Removals
- ❌ Removes reset after X failed WiFi reconnect attempts, because s [... cut]
commit 8b14a4c9 (2023-09-14): V0.86

### ✨ Features
- 🎵 Adds `/rtttl` API to play a rtttl melody string directly
- 🖥️ Adds HTTP fullscreen liveview at `/fullscreen`

### 🐛 Bug Fixes
- 🎨 Fixes Matrix color for HA entity. Closes #314
- 🌈 Fixes moodlight bug. Closes #311
- 📆 Fixes drawing coordinates for time calendar icon. Closes #312
- 🔡 Fixes a bug with the free pixel between icon and text
- 🟠 Fixes circle drawing
- 📝 Fixes a bug while using textOffset and pushIcon. Closes #305

### 🔥 Removals
- ❌ Removes reset after X failed WiFi reconnect attempts, because some users disables WiFi over night. Closes #310
commit d76c8a73 (2023-09-14): v0.86

### ✨ Features
- 🎵 Adds `/rtttl` API to play a rtttl melody string directly
- 🖥️ Adds HTTP fullscreen liveview at `/fullscreen`

### 🐛 Bug Fixes
- 🎨 Fixes Matrix color for HA entity. Closes #314
- 🌈 Fixes moodlight bug. Closes #311
- 📆 Fixes drawing coordinates for time calendar icon. Closes #312
- 🔡 Fixes a bug with the free pixel between icon and text
- 🟠 Fixes circle drawing
- 📝 Fixes a bug while using textOffset and pushIcon. Closes #305

### 🔥 Removals
- ❌ Removes reset after X failed WiFi reconnect attempts, because some users disables WiFi over night. Closes #310