Blueforcer/awtrix3 #321: [BUG] Custom App Name seems to be NULL

Gerrit-KMeier, 2023-09-15:
# Bug report

## Describe the bug

I wantet to create a custom app for solarpanel. So i POST this payload to the clock
```
{
    "name":"pvin",
    "text":"0W",
    "icon":"27464",
    "color":"#007700"
}
```
the result on the display is as expected and the icon is properly displayed. But this custom app seem not to have a name. I can not select it with

`http://{ip}/api/switch` and this payload
```json
{
    "name": "pvin"
}
```
MQTT displays this data

`awtrix_2/stats/currentApp (null)` 

when the custom app ist active.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.87

## To Reproduce

Post this payload to the endpoint `http://{ip}/api/custom`

```
{
    "name":"pvin",
    "text":"0W",
    "icon":"27464",
    "color":"#007700"
}
```

## Expected behavior

The custom app has the desired name »pvin«

## Screenshots

n/a

## Logs
Only the time and custom app are active and alter automatically.

```text
awtrix_2/stats/currentApp (null)
awtrix_2/stats/currentApp Time
awtrix_2/stats {"bat":100,"bat_raw":709,"type":0,"lux":16,"ldr_raw":654,"ram":146000,"bri":120,"temp":35,"hum":37,"uptime":1122,"wifi_signal":-51,"messages":0,"version":"0.87","indicator1":false,"indicator2":false,"indicator3":false,"app":"Time","uid":"awtrix_ed5c20"}
awtrix_2/stats/currentApp (null)
awtrix_2/stats {"bat":100,"bat_raw":678,"type":0,"lux":16,"ldr_raw":654,"ram":146036,"bri":120,"temp":35,"hum":37,"uptime":1132,"wifi_signal":-47,"messages":0,"version":"0.87","indicator1":false,"indicator2":false,"indicator3":false,"app":"","uid":"awtrix_ed5c20"}

```

## Additional context

Blueforcer, 2023-09-16:
That's not how it works.
You need to send the name as parameter in your http request.

Gerrit-KMeier, 2023-09-16:
ok you might think I mixed up MQTT and HTTP requests. But at first I perform all my first steps with http requests.

`http://{ip}/api/custom/pvin`
with POST payload
```
{
    "name":"pvin",
    "text":"0W",
    "icon":"27464",
    "color":"#007700"
}
```

does not work

and neither does
`http://{ip}/api/custom`
with a POST payload
```
{
    "name":"pvin",
    "text":"0W",
    "icon":"27464",
    "color":"#007700"
}
```
and calling it with
`http://{ip}/api/switch/pvin` → 404 Error

[image]

I'm using [RESTer Addon](https://addons.mozilla.org/de/firefox/addon/rester/) fpr FireFox to perform these tasks. And `{ip}` is a variable which contains the ip-address.

so if the app is supposed to be called pvin where do I need to place the parameter(s)? Btw I also can only declare one app. If I create another one as described above with a different name, the current is overwritten.

Blueforcer, 2023-09-16:
A http Parameter needs to be in the url:
http://{ip}/api/custom?name=pvin

Gerrit-KMeier, 2023-09-17:
This throws an 404 error
[image]

in the meantime I utilized MQTT and had no issues at all.
See this verbose output of the command `mosquitto_sub -d -v -t awtrix_2/#`
```
Client (null) received PUBLISH (d0, q0, r0, m0, 'awtrix_2/stats/currentApp', ... (4 bytes))
awtrix_2/stats/currentApp pvin
```

Blueforcer, 2023-09-17:
Because you need to send a POST request, not GET...

Blueforcer, 2023-09-18:
Is this solved?

closed 2023-10-14 by Blueforcer