Blueforcer/awtrix3 #324: [BUG]

DrRSatzteil, 2023-09-16:
# Bug report

## Describe the bug

Disabling automatic transition (ATRANS) via MQTT does not seem to work. After sending {"ATRANS":false} the apps keep changing anyway. Disabling the setting in the iOS App however works as expected.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)

## To Reproduce

Steps to reproduce the behavior:

Publish {"ATRANS":false} on ?/settings topic. Apps are still transitioning after that.

## Expected behavior

The currently active app should be shows until ATRANS has been enabled again.

## Screenshots

N/A

## Logs

N/A

## Additional context

N/A

Gerrit-KMeier, 2023-09-17:
I tried to reproduce your issue but failed. The prefix for my Device is "awtrix_2" so I entered
`mosquitto_pub -d -t awtrix_2/settings -m '{"ATRANS":false}'`

result:
```
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'awtrix_2/settings', ... (16 bytes))
Client (null) sending DISCONNECT
```

and the current active page does not change anymore. So check your version or if you have a type in your command.

DrRSatzteil, 2023-09-17:
Thank you for your input. I can confirm that it works as expected at least for the default apps.

However, when I have a custom app that is set to e.g. repeat = 3 the clock will switch to the next app after three repeats no matter if ATRANS is set to true or false. I must admit that it is not quite clear what the default behaviour would be in that case. I was just assuming that it would repeat infinitely when ATRANS is set to false.

Gerrit-KMeier, 2023-09-17:
yes I guess some options exclude each other on a logical base. One need to decide which option wins in favour to another. So when you forbid transition with `{"ATRANS":false}` should the property `{"repeat":3}` of the custom app be ignored or does this option property `{"repeat":3}` override the disabled transition.

In my humble option I would declare `ATRANS` as a global option with a higher priority. Properties bound to an app, which might be affected by a global setting, do therefore have a lower priority and are ignored/overridden.

That said, this might be worth a discussion

Blueforcer, 2023-09-17:
It has already been changed 2 back and forth in the past. At some point it is no longer possible to please everyone. So I do it like I prefer 😁

closed 2023-09-17 by Blueforcer

DrRSatzteil, 2023-09-17:
I also think it would make more sense to use this as an override for app specific settings. With the way it works now this I don't really see a use for the ATRANS setting at all because you cannot foresee what will happen when you disable it.

However its of course your choice and that is fine. I will try to experiment with a workaround by repeatedly switching to the current app, maybe this will work.

DrRSatzteil, 2023-09-17:
I know that this issue is closed already but just for completeness: I did some more testing and concluded that the repeat property is evil. It is not only problematic regard to the ATRANS setting but also when the displayed text is too short for the app duration (you get a black screen for the remaining time). So I just decided to not use repeat at all and this basically solved all my problems 😂