Blueforcer/awtrix3 #326: [FEATURE REQUEST] Add Deep Sleep for X amount of time

ejalal, 2023-09-18:
# Feature Request

### Is your feature request related to a problem?
Power endpoint only diables the LED Matrix, but stuff is still running in the background.

### Describe the solution / feature you'd like
It would be interesting to add a deep sleep endpoint to allow deep sleep for X amount of seconds received as a parameter (seconds calculated from Home assistant or whatever other app).

Could be acheived by:
1. getting a precalculated X amount of seconds (to be convertd to µs by awtrix), 
2. getting the exact time for wakeup HH:MM (more user friendly) and from there awtrix will convert it to remaining µs before HH:MM

For example:
In my automation (Home Assistant) I check for the "presence" of my desktop pc or laptop, and switch off the LED matrix AND a power outlet if none is detected (power outlet supplies the all my setup ulanzi matrix included)

Then I switch on the power outlet and led matrix in the morning at 8am (as a bonus this powers on the laptop which is nice for remote work), but some days the LED matrix doesn't power on as battery is fully discharged overnight (even with LED matrix off).

Deep sleep would allow me to force a deep sleep and wake up with a timer at 8am in my use case for example.

### Describe alternatives you've considered

A clear and concise description of any alternative solutions or features you've considered.

### Additional context

Add any other context or screenshots about the feature request here.

Blueforcer, 2023-09-18:
Currently I don't think it worth the effort for this specific case.
The current scaffold is not made for this.

ejalal, 2023-09-19:
I started working on this yesterday but had the ulanzi matrix reboot when I call esp_sleep_enable_timer_wakeup(xxxx) and esp_deep_sleep_start() so not getting anywhere right now, I might give it another try this evening.

Added an /api/sleep target and a new PowerManager.h/cpp class to handle this

IzeQube, 2023-09-20:
I actually like this idea. I also power off the sockets in my office where the ulanzi is connected if the office is not in use...

ejalal, 2023-09-23:
Well, this doesn't seem too stable as the ulanzi board wakes up randomly because it can't keep tack of time when put in deep sleep mode.

Given that it's put in deep sleep by home assistant with a precalculated number of seconds (driven by a sensor that keeps updating in home assistant), a workaround for this it to add some home assistant/nodered flow to put it back to sleep if it's online (ping device_tracker) before expected time with a freshly calculated number of seconds.

ejalal, 2023-09-27:
Last change in the PR I submitted seems to fix the early wakeup issue

closed 2023-10-15 by Blueforcer