Blueforcer/awtrix3 #329: [BUG] Current licensing prevents commercial use altogether

flobeier, 2023-09-20:
# Bug report

## Describe the bug
The README states to contact you if one wants to use the project commercially. However, since it contains code from contributors and you're not using a [Contributor License Agreement (CLA)](https://github.com/cla-assistant/cla-assistant), it wouldn't be possible to do so without asking every contributor for permission.

Blueforcer, 2023-09-20:
So what license I need then?

flobeier, 2023-09-20:
The issue is that the contributors retain full rights to their code as no CLA is in place. This means you can't change the license or distribute the code under a different license without asking them for permission or removing their code. Since the license for their contributions is the same as the license for the project, no commercial use would be possible until done so.

Blueforcer, 2023-09-20:
I can ask them, no problem.
Then I will add the CLA

Blueforcer, 2023-12-23:
commercial note removed

closed 2023-12-23 by Blueforcer