Blueforcer/awtrix3 #332: [BUG] Lametric icon download broken for jpg

TheGroundZero, 2023-09-22:
# Bug report

## Describe the bug

JPG (static) icons downloaded via the download functionality on the device appears to be broken.
Downloading GIFs works fine.

I did notice that the preview of the icon (which does function) displays a PNG, so perhaps there's a conversion error occurring?

## To Reproduce

Steps to reproduce the behavior:

- Preview a static icon from Lametric, e.g. `32047`, notice that it appears correctly
- Download this icon
- Check the icon in the File Manager, notice how the icon is broken
- Use the icon in a custom app, e.g. `{ "text": "test", "icon": "32047"}`, notice how it appears the same (broken) on the device

## Expected behavior

A clear and concise description of what you expected to happen.

- Download a static icon and have it appears correctly

## Screenshots

If applicable, add screenshots to help explain your problem.

[image]
[image]

Blueforcer, 2023-09-22:
It's just a regular download of the file. No converting happening

TheGroundZero, 2023-09-22:
> It's just a regular download of the file. No converting happening

Could it be that it's downloading a PNG but storing as a JPG?
The preview on the web interface is a PNG file. Also, renaming one of the broken jpg's to png does "fix" it in the file browser.

TheGroundZero, 2023-09-22:
I've used the preview feature to download the PNG of each icon, then used PaintDotNet to convert them into JPG.
Also renamed them for my own clarity.

For some reason, I'm not getting an icon on my device when I'm calling for a JPG icon. GIF icons work fine.
Text and progress bar display fine. Text is moved to the right to make room for an icon. But icon is black.

[image]
[image]

Calling for a GIF icon works fine
[image]

Blueforcer, 2023-09-22:
Does it have s resolution of 8x8?

TheGroundZero, 2023-09-22:
> Does it have s resolution of 8x8?

Yes. It's the PNG I downloaded from the preview and converted into JPG
[image]

bat_dch_100.jpg
[image]

bat_stb_full.gif
[image]

Blueforcer, 2023-10-14:
just tested it with 32047 and it works. preview in filemanager ist correct, also on matrix

closed 2023-10-25 by Blueforcer