Blueforcer/awtrix3 #334: Stuck Notification

satyambaba, 2023-09-24:
# Bug report

## Describe the bug

Once I send a notification with hold: true in the payload, then I dismiss the notification. After this, any notification I send gets stuck indefinitely even with hold: false and duration: 10. I have to manually dismiss these notifications. The intended behavior is that the notification should be displayed only for 10 seconds if duration is 10 and hold is false. Once I reboot, the issue gets resolved.

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.88

## To Reproduce

1. Send a notification with hold: true in the payload
2. Dismiss the notification
3. Send a notification with hold: false and duration: 10

## Expected behavior

The intended behavior is that the notification should be displayed only for 10 seconds if duration is 10 and hold is false.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-10-15:
i catn reproduces it, works like it should

satyambaba, 2023-10-17:
> i catn reproduces it, works like it should

Thanks for your response. It is still happening with me. Please let me know if you need any additional info in order to understand the issue better.

Blueforcer, 2024-03-17:
Back to this issue.
Does it still exist in the latest firmware?

satyambaba, 2024-03-18:
> Back to this issue. Does it still exist in the latest firmware?

It has been resolved in the latest firmware, you may close this issue. Thanks for your help.

closed 2024-03-18 by Blueforcer