Blueforcer/awtrix3 #335: Custom app not removed after time set through "lifetime" value

Kefut, 2023-09-25:
Hi,

I'm using a Ulanzi TC001 with AWTRIX-LIGHT v0.48 and publish custom apps through MQTT from Homeassistant.
As you can see from the snippet below I try to delete the custom app after 24h (86400 seconds) in case there has not been an update. Unfortunately the "lifetime" value doesn't seem to work as the app just doesn't disappear.

`service: mqtt.publish
  data:
     qos: 0
     retain: false
     topic: awtrix_1/custom/batwarn
     payload: >-
        { "text": "{{sensors}}", "icon": 12123, "textCase": 2, "save": true,
         "lifetime": 86400, "scrollSpeed": 75, "repeat": 1}`

Is this a bug or am I doing something wrong here?

Thanks for your help

Blueforcer, 2023-09-25:
Don't open a issue with such an old firmware version. We're at 0.86 ...

Kefut, 2023-09-25:
My mistake - I am actually on 0.88! 0.48 was a typo. Sorry for the confusion.

Blueforcer, 2023-09-25:
You need to set the lifetime in seconds.
Not milliseconds

Kefut, 2023-09-25:
I want 24h, so it's 24*60*60 --> 86400sec

Blueforcer, 2023-09-25:
I'm not sure it's made for such a long period ..
Need to check it this afternoon.

Kefut, 2023-09-25:
Thanks for checking - didn't see any limitations in the docu besides that the number needs to be an integer and didn't expect 1day to be too long.

Blueforcer, 2023-10-15:
Should be fine now by change the datatype to long. 
I didnt test it with 24h, please feel free to reopen the issue if the problem still persist

closed 2023-10-15 by Blueforcer

Kefut, 2023-10-24:
Just checked and unfortunately it's still not working.

heimchemiker, 2024-07-11:
I also have this problem. I'm using OpenHAB to publish this data to my MQTT Server, but this should not make a difference. I want it to not show the solar production one hour (=3600s) after it turned to "producing" 0 W. Here my last MQTT publish is from 21:47 (it's now 2:13, so this last change was ~4.5h ago), the MPTT Message was:
`{
  "icon": "sol-stop",
  "text": "0 W",
  "duration": 5,
  "color": "#FFFFFF",
  "progress": 0,
  "progressC": "#FFFFFF",
  "progressBC": "#222222",
  "lifetime": 3600
}`
And it was still displaying the message.
I'm on version 0.96