Blueforcer/awtrix3 #336: [BUG] Icon width not displayed correctly

k-a-r-g, 2023-09-26:
# Bug report

## Describe the bug

All icons in custom apps and notifications seem to be displayed with 8 pix width. Stock apps display the full width.
Issue can be observed best with calendar icons from lametrix.com, e.g. id #11567.
The icon is also displayed smaller in the file section of the http server.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version:  v0.88 

## To Reproduce

`
service: mqtt.publish
data:
  qos: 0
  retain: false
  topic: awtrix_kitchen/notify
  payload: |-
    {
      "text": "Test",
      "icon": "11567"
    }
`

## Expected behavior

Display of full icon.

## Screenshots

https://github.com/Blueforcer/awtrix-light/assets/11861332/c7a60251-55cc-4e09-b6d5-a596a4c65398

Blueforcer, 2023-10-11:
All icons are 8x8. also ID 11567

k-a-r-g, 2023-10-11:
Hi, thanks for the reply.
My bad, I guess it is not a bug then and can be closed.
As a feature request: is it possible to add an offset for scrolling text to the JSON properties?
Like in combination with pushIcon=0, so that a particlular part of the screen is left untouched?

Blueforcer, 2023-10-11:
There is already a textOffset option

k-a-r-g, 2023-10-12:
I feel so bad wasting your time.
Anyway, thanks for your great support for a great application!

closed 2023-10-12 by Blueforcer