Blueforcer/awtrix3 #337: [BUG]

10der, 2023-09-26:
# Bug report

## Describe the bug
switch not is case-sensitive

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.88 

## To Reproduce

Steps to reproduce the behaviour:

rtfm `{"name":"time"}`
- send to mqtt
- nothing
change 
`{"name":"Time"}`
now all ok

## Expected behaviour
idk how right :-) 
please fix or code or rtfm =) 

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-09-26:
I'm not sure what's the big here?

zzarne, 2023-09-27:
I've stumbled there as well when I was reading the API documentation. The example for /switch has "time" as an example while further on the built-in names are "Time", "Date" and so on. "time" != "Time". I agree that this is mislabeled as a bug, but the documentation could me more precise and state which topics and payloads are case sensitive and which aren't.

Blueforcer, 2023-09-27:
Thats just an example. Customapps can be whatever you like
The names of the build-.in apps are listed below (see your own screenshot)

closed 2023-09-27 by Blueforcer