Blueforcer/awtrix3 #339: [BUG] Dots seperator of time stoped blinking

MHPtweidinger, 2023-09-29:
# Bug report

Dots of the clock app don't flash after changing the config with MQTT.

## Describe the bug

The dots separator between hours and minutes of the default clock app used to blink. Today i submitted a MQTT config to adjust the color and brights and played around with the formatting of the clock app and as i published my first config the dots stopped flashing. I looked in the docu but could not find a property to turn the flashing back on. 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

Blueforcer, 2023-09-29:
What is your current timefomat

MHPtweidinger, 2023-09-29:
i set it to: 
`"TFORMAT":"%H:%M",
`

Blueforcer, 2023-09-29:
Replace : with a space

Blueforcer, 2023-09-29:
See documentation:
[image]

closed 2023-09-29 by MHPtweidinger