Blueforcer/awtrix3 #341: [BUG] Flashing my Ulanzi clock doesn

Jonas-Sander, 2023-10-01:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Jonas-Sander, 2023-10-02:
Okay, I solved by connecting to the Ulanzi clock via the IP address shown on the device, then tapping "About" (the button at the very bottom of the page), selecting update firmware, and choosing "ulanzi_TC001_0.88.bin" that is at the "releases" tab here on GitHub. Since I already added the wifi before updating the clock it connected directly to my wifi after flashing it with the AWTRIX firmware.

Jonas-Sander, 2023-10-02:
This process is not documented though and the original way (the quickstart at https://blueforcer.github.io/awtrix-light/#/quickstart) didn't work for me.

Blueforcer, 2023-10-02:
You need to use a high quality cable and probably another USB port. The ulanzi hardware is bad.
But glad it works now.

closed 2023-10-02 by Blueforcer

Jonas-Sander, 2023-10-03:
> You need to use a high quality cable and probably another USB port. The ulanzi hardware is bad.
But glad it works now.

I tried it with 2 other USB cables and 2 more USB ports, still didn't work :/

Blueforcer, 2023-10-03:
Sorry there is nothing I can do.
Some users reported that it works on another PC for any reason.

Jonas-Sander, 2023-10-03:
No worries, thanks for your help :)   

There is one thing that was annoying though because of this issue: When I had issues with the IP settings (I changed something and couldn't connect to the web interface afterwards) I had troubles resetting the device. The reset button at the back of the device didn't function properly. If you press it nothing happens. I tried pressing, holding, turning the device on and off several times etc. and it didn't work. After a while I got lucky and it somehow went back to the access point mode. I think it was when pressing the reset button while the device is booting up? I tried to reset it again afterwards once and it didn't work. Is there a special way or time when the reset button has to be pressed?

pzzz, 2023-10-27:
> Okay, I solved by connecting to the Ulanzi clock via the IP address shown on the device, then tapping "About" (the button at the very bottom of the page), selecting update firmware, and choosing [...]

Can confirm, this also was the way to go for me (y)

GitGudin, 2023-12-12:
> Okay, I solved by connecting to the Ulanzi clock via the IP address shown on the device, then tapping "About" (the button at the very bottom of the page), selecting update firmware, and choosing [...]

Worked for me, too. Thanks for sharing!

runrennerrun, 2023-12-27:
> Worked for me, too. Thanks for sharing!

Hi, I can't find the  releases tab where the "ulanzi_TC001_0.88.bin" should be. where do i find that?

coalfield, 2024-01-22:
Thank you, really helped me after trying numerous cables!

wAuner, 2024-02-03:
I can confirm that using a better cable helped. When I switched to a very short (about 50cm) Thunderbolt 4 cable I was able to successfully flash the firmware with my Macbook.

mbnn6, 2024-02-22:
I would like to confirm that changing the cable helped! It's because of the way the clock will connect to your mac, choosing serial connection as the connection method (important to make it work, don't choose bluetooth) was only possible then.

SimonKai22, 2024-03-14:
> ulanzi_TC001_0.88.bin

I have the same problem. Can you tell me where I can find the file "ulanzi_TC001_0.88.bin"?

many thanks for the help

Blueforcer, 2024-03-14:
> Many thanks for the link. If I have managed to install this version on the Ulanzi, what do I have to do afterwards?

What do you mean? I think there must be a reason why you're flashing awtrix.
You should read the documentation for further steps

SimonKai22, 2024-03-14:
oh ok, I think I misunderstood that. Is "ulanzi_TC001_0.88.bin" the same version as the one via the web flasher?

SimonKai22, 2024-03-14:
That worked! Thank you
How do I switch off the temperature and humidity display? Unfortunately I don't understand the instructions

noizee2k, 2024-03-24:
Yep. It worked that way, had the same problems on my mac. Already bought your app, worth the money!

MeisterPiffer, 2024-04-01:
I just want to add that I had the same issue.
I even flashed the firmware via commandline esptool, but even though reporting success, it didn't came up again.
However I was able to reflash the original firmware and use the about->update method to flash the very same file I tried before and this worked.
I feel like there should be a hint to this procedure as alternative to the Online Flasher, as I tested several combinations including 3x USB A to C cables, 3x USB C to C cables, 2x Linux computers, one Windows Laptop and 2 TC001 without success.

Jonas-Sander, 2024-04-01:
I also think this should be highlighted / documented on the [flasher](https://blueforcer.github.io/awtrix3/#/flasher) site.

PtiDam27, 2024-04-17:
> Okay, I solved by connecting to the Ulanzi clock via the IP address shown on the device, then tapping "About" (the button at the very bottom of the page), selecting update firmware, and choosing [...]

this work for me too big thank i try a lot solution i found on internet this solution only work and is so easy

Blueforcer, 2024-04-17:
Flashing without changing the partitions, will results in less free Flash space. Sou you have less space available for Icons etc.

derdritte, 2024-04-29:
I also cannot use the online flasher (ubuntu 22.04, chrome 124.0.6367.91). For safety I wanted to read/dump the base-firmware first using esptools, which initially also failed. Only after I removed all baud-settings, and dumped using `esptool.py -c esp32 -p /dev/ttyUSB0 read_flash 0x0 0x400000 ulanzi_tc001_factory.bin` would it work.

I only tried on the one PC so far, but I tried 3 cables, including a usb-c/usb-c one none of which worked for the online flasher, but did work with esptools.

My ulanzi just reset at the very beginning of the flashing-process (showing the logo in red again) and then just acted like it was reset. I would love a second route using locally installed esptools (in addition to the built-in online flashing, which other users already mentioned).

Raudi1, 2024-05-25:
I had the same issue. Tried all kinds of different cables (C to C, A to C, USB 2/3, short, long...) on different ports connected to different controllers on my computer (has no USB2 though).
What worked for me immediately was to connect through the USB3 hub built into my monitor.

Blueforcer, 2024-06-17:
> What worked in my case was to flash the firmware through the built-in webinterface of the clock. The online-flasher just didn't come through on my computer.

You will loose lots of flashmemory with this method, becaue ulanzi use a different partition table.

dhuddle, 2024-06-19:
> I also think this should be highlighted / documented on the [flasher](https://blueforcer.github.io/awtrix3/#/flasher) site.

This bug report and resolution helped me with my issue. 4 different USB cables, 5 different computer ports, 2 different browsers (chrome and edge)  and no luck. Once I went to the ulanzi stock firmware update and hit enter, it was cake.

It should at least be documented on the official flasher site as an alternative method.

overtesters, 2024-06-23:
> I had the same issue, different USB cables und ports didn't work.
> 
> What worked in my case was to flash the firmware through the built-in webinterface of the clock. The online-flasher just didn't come through on my computer.

A little update to my original post: After Blueforcer's Post I continued to try getting Awtrix 3 on my Ulanzi with the online Flasher. Installing through my Ubuntu PC didn't work either. In the End it worked with a Barebone PC where a minimal Installation of Windows 10 was running. Took about 2 minutes, only strange thing was that the Ulanzi was beeping through the whole update process. Aside from that, it worked liked described in the online manual.

Blueforcer, 2024-06-23:
Beeping while flashing is normal because it's a active low buzzer.

iamjackg, 2024-11-02:
Just wanted to add that I was running into the same issue, but flashing locally with esptool following the instructions at https://github.com/Blueforcer/awtrix3/blob/main/docs/ulanzi_flasher/firmware/README worked fine.

shinz41, 2024-11-10:
On linux,  you might need permissions to access a device/port :
sudo setfacl -m u:USERNAME:rw /dev/ttyUSB0
Where your session account name should replace USERNAME (of course...)

Olli2k, 2024-12-03:
0Same Issue here. Depends on the USB3 technology of "modern" PC. Using a USB Hub with 2.0 Standard solved this for me. Should work for everyone else. Tested it on all PCs and with all cables I used before. All were working with the USB Hub.

dmxtoxlr, 2025-01-19:
> 0Same Issue here. Depends on the USB3 technology of "modern" PC. Using a USB Hub with 2.0 Standard solved this for me. Should work for everyone else. Tested it on all PCs and with all cables I used before. All were working with the USB Hub.
> 
> €dit: All tests were done with a already flashed awtrix3 (0.96) via webinterface.

Thanks so much for this hint! Since aparently we don't have any PCs with USB2.0 ports anymore, I ended up trying with an old raspberry pi3. Worked perfectly fine with (with the chromium browser) on the first try :)
Maybe you could add this information to the quick start guide?

gfx54, 2025-08-05:
The solution provided up (upgrading using the "About" button + firmware ulanzi_TC001_0.98.bin  worked well on my Ulanzi (bought in July 25).

Blueforcer, 2025-08-05:
> The solution provided up (upgrading using the "About" button + firmware ulanzi_TC001_0.98.bin  worked well on my Ulanzi (bought in July 25).

Please note, with this solution you just have 250kb of free flash instead of 1,35MB

cooper2k4, 2025-08-12:
Using an USB 2.0 Port instead of 3.0 solved that issue for me.

Nemeczek, 2026-02-02:
I read somewhere that if you have it error then try USB 2.0 port. And it worked!

rok8ray, 2026-05-09:
Hey, i'm having this same error. Could someone please give me the link to the releases tab you speak of? Thanks