Blueforcer/awtrix3 #344: Icon Database Webinterface

hallerdavid, 2023-10-09:
Hello togehter, i cant find the icon database in my awtrix light webinterface.. where can i import lametric icons? 
i have awtrix light 0.88 on my ulanzi.

closed 2023-10-09 by hallerdavid

udo1toni, 2023-10-09:
Go to https://developer.lametric.com/icons and find your icon. write down its number and go to awtrix light incos page. type in the id of the lametric icon.

mc-0815, 2024-01-21:
Actually my webinterface (version 0.94) does not have an icon page. Where can the awtrix-light icons page be found?

hallerdavid, 2024-01-21:
try another browser :-) that solved my problem

mc-0815, 2024-01-21:
Tried firefox as well as chrome (both on Ubuntu) with same result only 

- WiFi Setup 
- Files 
- Update 
- LiveView 
- Backup 
- Docs 
- Flows

are presented, see html source

`<div class=topnav id=top-nav>
                <span id="nav-link">
                    <a id=set-wifi class="a-link active" data-box="wifi-box">WiFi Setup</a>
                </span>
                <a id=about target=_blank rel=noopener href=[./edit](view-source:http://127.0.0.1:10080/edit)>Files</a>
                <a id=about target=_blank rel=noopener href=[./update](view-source:http://127.0.0.1:10080/update)>Update</a>
                <a id=about target=_blank rel=noopener href=[./screen](view-source:http://127.0.0.1:10080/screen)>LiveView</a>
                <a id=about target=_blank rel=noopener href=[./backup](view-source:http://127.0.0.1:10080/backup)>Backup</a>
                <a id=about target=_blank rel=noopener href=[https://blueforcer.github.io/awtrix-light](view-source:https://blueforcer.github.io/awtrix-light)>Docs</a>
                <a id=about target=_blank rel=noopener href=[https://flows.blueforcer.de](view-source:https://flows.blueforcer.de/)>Flows</a>
                <a class="a-link icon" id=hum-btn>
                    <div class=svg style=margin:-2px>
                        <svg id=svg-menu fill=currentColor viewBox="0 0 20 18"></svg>
                    </div>
                </a>
            </div>
`

Blueforcer, 2024-01-21:
Delete the donottouch.json and restart awtrix.
This will reset the Webinterface including the settings you had made there like Matt or time (except of wifi and filesystem etc)

mc-0815, 2024-01-21:
still the same issue, webinterface does not show other configuration options or icon downloader

mc-0815, 2024-01-22:
is there any opportunity to debug this behaviour? Device is connected to my WLAN and got IP from DHCP server however also the WiFi setup page does not show any SSID nor IP. So it might the icon issue might be connected?!

Thanks

Blueforcer, 2024-01-22:
Open developer console on your browser.
Also make sure you doesn't run any browser extension wich could block something

mc-0815, 2024-01-22:
Thanks, found the issue with the ntwork console. 

I operate my IoT devices in a separated network that is not routed. So I need to tunnel the web ports via ssh like ```ssh -L 10080:<awtrix ip>:80 <IP>```. However the code in the webpage does not respect the port (in my case 10080) when trying to access DoNotTouch.json via ```GET http://127.0.0.1/DoNotTouch.json```. When I change the port to standard like ```sudo ssh -L 80:<awtrix ip>:80 <IP>``` website works.