Blueforcer/awtrix3 #345: [BUG] temp_dec_places in mqtt message _tmp

martin131, 2023-10-11:
# Bug report

## Describe the bug

temp_dec_places is not reflected in _tmp message in mqtt

## Additional information

- Devices involved:
  - Model: ULANZI with DIY ESP32
  - AWTRIX-LIGHT version: 0.88

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

[image]

Gerrit-KMeier, 2023-10-12:
The temperaturesensor in the device is quite inaccurate. So do you utilize it somewhere ?

martin131, 2023-10-13:
@Gerrit-KMeier I am using DIY ESP32 setup with external connected SHT31 sensor. Now it is very accurate.

closed 2023-10-15 by Blueforcer