Blueforcer/awtrix3 #347: [BUG] Progress bar left alignment

awolkers, 2023-10-12:
# Bug report

## Describe the bug

The width of the progress bar when there is an icon active does not align with the left side of the text scroll area. It would be more logical that the width of the progress bar is the same as the width of the text scroll area. The progress bar now 'sticks' to the right side of the icon. 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.88 ]

## To Reproduce

Add a custom app with an icon and text that exceeds the size of the scroll area.

## Expected behavior

Remove the first pixel and make the progress bar as wide as the text scroll area.

## Screenshots

[image]

Blueforcer, 2023-10-12:
thats not a bug. The shorter the progressbar is the smaller is the resolution to display 0-100%. That is why every free pixel is used to

awolkers, 2023-10-12:
Okay I get the idea of making the bar a wide as possible but don't get it why there is no space between the icon and the bar. For the text there a column of 'whitespace' reserved but not for the progress bar. Take for instance this example where the progress bar is sticking to the icon. It would be more aesthetically pleasing if the bar also respects this whitespace.
 
[image]

Blueforcer, 2023-10-12:
> Okay I get the idea of making the bar a wide as possible but don't get it why there is no space between the icon and the bar. 

Exactly for this reason.The majority wanted it that way at the time, so I implemented it that way. If you want to discuss with the other users again, then you can join the discord server, and start there a discussion about it.

closed 2023-10-12 by awolkers