Blueforcer/awtrix3 #349: When sending Post for custom app, an effect is set

SKB-CGN, 2023-10-15:
Hi,
I am sending apps to the clock via Post. This was working well. Today I updated the clock and now there is some effect behind my text, which I do not want.

The effect value is set to empty or 'None' but displays a color waving background.

How can I have the behaviour like before?
How do I unset the background effect?

'none', 'None', false are not working.

Thank you!

Blueforcer, 2023-10-15:
If you don't need aneffect, don't send the key "effect". A effect "none" doesn't exist and awtrix uses the first one

SKB-CGN, 2023-10-15:
Before I sent empty key effect, which was working fine. Means, if no key effect ist sent, the effect will not displayed?

Blueforcer, 2023-10-15:
But this has changed. Why you send a empty value if you don't use it?

SKB-CGN, 2023-10-15:
I create the apps dynamically - therefore the key exists and got a value or was set to empty. I will need to change that.

Thanks.

closed 2023-10-15 by SKB-CGN

Blueforcer, 2023-10-15:
I will see if I can do another solution..

SKB-CGN, 2023-10-15:
Thanks a lot!

SKB-CGN, 2023-11-05:
Did you change it back again that if effect is empty, that the effect is not applied?

Just to be sure, because I installed the new version and the effect is gone 👍