Blueforcer/awtrix3 #350: [BUG] Automatic brightness too bright (100%) in dark environment

dYn36, 2023-10-15:
# Bug report

## Describe the bug

After I upgraded to version 0.89, the Auto brightness does not work as before. The LEDs are at 100% when the environment is about 6 lux.
Likewise, if the environment is 10,000 lux, the LEDs are also at 100% (the app at least shows 100, so I think it means percent).
In relation to the environment, it even dazzles a bit (if you can say that).

The previous version did that better.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.89

## To Reproduce

Steps to reproduce the behavior:
Update to v0.89 and activate „Automatic Brightness“. See that the brightness is too bright in relation to the Environment.

## Expected behavior

Auto brightness should work Like in the former version v.0.88.
Alternatively, it would be good to be able to adjust the sensitivity of the Auto Brightness function.

Blueforcer, 2023-10-15:
I made some adjustments in 0.90.
You can also set the calculation range via dev.json to fit your needs.

dYn36, 2023-10-15:
I think with Version 0.90 its better. Thanks!

closed 2023-10-16 by Blueforcer

mapietru, 2023-10-24:
Hello, I have just updated to 0.90 and having the same problem. The brightness level is far to high in low light conditions. To be honest I was very happy with the implementation prior 0.89...

SmartHomeFanatic, 2023-10-24:
@mapietru I solved this with a home assistant automation now. Everytime the illuminance value changes I adjust the brightness. That makes it even more flexible.

mapietru, 2023-10-25:
Hello, sure thats one way so solve it. But why should I use a script to control the brightness? The auto brightness was working well for me. Better solution would be to have some table with lets say 15 by 2 to build pairs of sensor input vs. display output. Just let the device do this job.

SmartHomeFanatic, 2023-10-25:
I know what you mean. Just wanted to give you an option so "solve" it yourself.

SmartHomeFanatic, 2023-11-06:
@speedAmaster sure:

`alias: AWTRIXs Brightness
trigger:
  - platform: state
    entity_id:
      - sensor.awtrix_0a8fac_illuminance
    to: null
    for:
      hours: 0
      minutes: 0
      seconds: 10
action:
  - if:
      - condition: state
        entity_id: light.awtrix_0a8fac_matrix
        state: "on"
      - condition: numeric_state
        entity_id: sensor.awtrix_0a8fac_illuminance
        above: 0
    then:
      - service: light.turn_on
        data:
          brightness_pct: "{{ (states('sensor.awtrix_0a8fac_illuminance') | float) * 4 + 1 }}"
        target:
          entity_id: light.awtrix_0a8fac_matrix`