Blueforcer/awtrix3 #352: [BUG] Fixed IP config not working?

DJF3, 2023-10-16:
# Bug report

## Describe the bug

v0.88 - after configuring Static IP under "Network", the clock doesn't use a static IP. It only listens to my DHCP server.
Configured: local IP, gateway, subnet (255,255,255,0) and primary DNS (also a local device).

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [0.88]

## To Reproduce

Steps to reproduce the behavior:
- in a terminal window (or equivalent) ping the current clock IP "ping 192.168.4.118 -a" (-a = beep on success)
- Enable Static IP (toggle goes green)
- Configure a NEW static IP address
- click "Save configuration"
- click "Restart ESP"

## Expected behavior
The ping should stop beeping because the clock has a new (fixed) IP address. Instead it doesn't.

Manually powering off/on the clock doesn't make a change either.

Blueforcer, 2023-10-18:
im using it since beginning, and it works as it should.
[image]

tebra-jl, 2023-10-26:
I have the same bug on same hardware with firmware 0.90

tebra-jl, 2023-10-26:
It works now after define also a secondary DNS and a reboot.
Thanks for this great software.

closed 2023-12-18 by Blueforcer