Blueforcer/awtrix3 #353: [BUG] HA MQTT discovery sensor 'Current App' is kept at last custom app when none is running anymore

pplucky, 2023-10-16:
# Bug report

## Describe the bug

HA MQTT discovery sensor 'Current App' is always kept at last custom app, when none in running anymore

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.90

Only 'Time' standard app is active in settings, no other.

## To Reproduce

Steps to reproduce the behavior:
- Create a custom app ABC (while this is active, 'Current App' switches between ABC and Time):

  Send to MQTT topic `awtrix_garagem/custom/ABC`, a payload like the following:
  ```
  {
    "text": "ABC test",
    "icon": "abc",
    "duration": 15
  }
  ```
- Remove the custom app ('Current App' is kept as ABC):

  Send to MQTT topic `awtrix_garagem/custom/ABC`, an empty payload

## Expected behavior

If no custom app is running, the 'Current App' should be set as 'Time' (or None?)

## Screenshots

Sensor value:
[image]

Sensor definition:
[image]

MQTT values:
[image]

MQTT stats values
[image]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

madcow27, 2024-02-21:
Hello,

I can confirm the same issue with the notifications.
Version 0.94.

If you send a notification for example with the "playground" function of the Android app, with the hold option, if you dismiss this notification then the "current app" value will stay on "Notification" unless you use the "next app" or "previous app" function (in my case I have usually only 1 app).

ShadowRep, 2024-05-16:
I have the same problem. I just got my TC001 and I thought I did something wrong.
This is really disrupting my experience with Awtrix 3.
I even deleted the device from HA and reset it. It always starts displaying the completely unwanted PlasmaCloud app in a loop.

iamjackg, 2024-11-13:
> if you dismiss this notification then the "current app" value will stay on "Notification"

This is probably because [`DisplayManager_::generateNotification`](https://github.com/Blueforcer/awtrix3/blob/6ec015b95b67758c22ae6a35d58f6924f1f5678c/src/DisplayManager.cpp#L970) calls `MQTTManager.setCurrentApp(CURRENT_APP)` but `CURRENT_APP` is never set back to the original value once the notifications are done rendering and `MQTTManager.setCurrentApp` is not called again until another app transition.

@Blueforcer would you mind taking a look when you have a moment? It's not a huge issue for me, so no rush.