Blueforcer/awtrix3 #354: [FEATURE REQUEST] Don't send zero temperature after reboot

martin131, 2023-10-16:
# Feature Request

### Is your feature request related to a problem?

If so, then add a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

### Describe the solution / feature you'd like

After every(?) reboot Awtrix sends zero temperature to MQTT. Is it possible to send any other value instead? Like Not available?

### Describe alternatives you've considered

As a workaround, I can use fitler in Home Assistant to filter these values.
### Additional context

[image]

luebbe, 2023-10-16:
It would be sufficient to send no value at after reboot until the values are stable. 
* Maybe the temp sensor needs a little "warm up" time, so delaying the temp/humidity reads for a few seconds could help.
* Or maybe the initial zero values are sent before the sensor has been read for the first time?

luebbe, 2023-11-12:
The same is also true for the battery levels. Here is a chart from MQTT explorer

[image]

you can see that after a reboot, first a value of zero and immediately afterwards a value of >2400 is reported, before awtrix settles on the correct value of about 600. Is this a reading error during the startup phase or an intialization error?
If it is not an initialization error, but a reading error, maybe you could implement a one-dimensional median filter to get rid of the peaks?

Blueforcer, 2023-11-12:
Many things are possible.
The question is, does it worth the effort?
Currently I don't see any downside of this behavior.

luebbe, 2023-11-12:
Looking at the code I think that one problem is that the sensor values may not have been read yet, when they are sent for the first time. Both reading and sending interval are set to 10 seconds by default.
So in order to get rid of the zero values it may be sufficient to delay the first call to sendstats until batt/temp/hum/lux have been read at least once. Another option would be to set the first reading interval of the sensor values to less than the sending interval for the first read and then back to 10 seconds after the first read.
I'll check this out and provide a PR if it works.

martin131, 2023-11-12:
@Blueforcer  In my setup the problem is that I am controlling thermostat with the temperature from awtrix. If the temperature drops to zero, my thermostat turns on.

luebbe, 2023-11-16:
The first part is done (no PR yet). Zero values are not sent anymore. Here's the graph (from LTR: batt_raw batt_%, temp, hum, ldr_raw). The first two peaks in each graph are reboots with the 0.90 version. The third reboot with my reworked version shows only a peak in batt_raw (high) and ldr_raw (low). The zero values are already sorted out.

[image]

I moved the Ulanzi on the second reboot, that's why the ldr_raw value changes (and probably also temp/hum start to creep, because the brightness went up).
Over the weekend I'll try to find time to sort out the peaks in batt_raw and ldr_raw and will provide a PR

closed 2023-11-28 by Blueforcer