Blueforcer/awtrix3 #357: update to 0.9 and sensor home assistant display is blank now.[BUG]

GL640, 2023-10-17:
i update to the latest firmware today and on reboot en reenter the mqqt i only get a blank screen when data is send from home assistant to the clock.
before the update i could send in a automation every 30 sec data from a power sensor but now after the update the clock stil response but with a blank screen
my older 0,59 version clock ( so not updated yet ) stil works with the same automation.

what happend with the new version that the mqqt data is not the same anymore?

# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-10-17:
Nothing has changed in 0.90 to the previous one with CustomApps. 
If you also upgrade this clock from 0.59 to 0.90 then there are of course lot of breaking changes.

GL640, 2023-10-17:
upgraded this from 0.7 to 0.9 ( the other was my first and stil runnig strong on 0.59 ) and with 0.7 it worked. even wy would the mqqt message data be changed thats a standard right? where there breaking changes?

Blueforcer, 2023-10-17:
I cannot tell you every changes in the last 20 version you have missed 😅 Check the changelogs and documentation.

But your JSON looks correct. Replace your entity state with a static text and check if the problem still exists. If not then your state is simply empty

GL640, 2023-10-17:
> I cannot tell you every changes in the last 20 version you have missed 😅 Check the changelogs and documentation.
> 
> But your JSON looks correct. Replace your entity state with a static text and check if the problem still exists. If not then your state is simply empty
ok i will check but the same data ( constant changing data ) is displayed on the other clock with 0.59 so its not empty ;-)
i can see if just displays text ... now only have to figure out again what that code was ;-)

--- wil try this ;-) ---
payload: "{ "text": {{ TEXT? }}, "duration": 5 }"

Blueforcer, 2023-10-17:
No
{ "text": "TEST", "duration": 5 }

GL640, 2023-10-17:
> No { "text": "TEST", "duration": 5 }

to see wy this happens i went to develepment tools and mqqt publis
--
service: mqtt.publish
data:
  qos: 0
  retain: false
  topic: awtrix_122264/notify
  payload: "{ "text": "TEST", "duration": 5 }"
--

and got a error "Service YAML contains syntax errors, please fix the syntax"
what wrong with the code?

GL640, 2023-10-17:
can i not just downgrade to 0.59 ;-) a lot easier haha

closed 2023-10-18 by Blueforcer