Blueforcer/awtrix3 #362: [BUG] Some icon could not be downloaded  "This ID doesnt exist"

servicemensch, 2023-10-20:
# Bug report

## Describe the bug

Some LaMetric icons could not be downloaded, even if preview is working

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.88

## To Reproduce

Go to web interface / icons
enter icon ID 6427, or 160 or 72
click preview
click download

## Expected behavior

icon should be downloaded to local storage, but this is not working for every icon in lametric database

## Error message

Error: This ID doesnt exist

Blueforcer, 2023-10-20:
I tested all 3 and it works fine. Try clearing your browsers cache.

GM-Jokemaster, 2023-10-21:
Same here 0.90 (different browsers same result) = This ID does not exist.
Download/Save of the Preview from the shown Preview as Animated gif works. (Right click Preview Picture save to)
Then i have to use the Files tab and upload manual. (If i klick ICONS first the folder icon shows as open wihout files (First time use its ok). The Path in the top does not change to ICONS main path.

If i choose manual /ICONS/yxz.gif the first time with no files in ICONS AWTRIX shows error. (can't reproduce) 
I now klick an icon under ICONS and the path changes correctly.
So upload in the root path works after that i have to right click move then enter /ICONS/yxz.gif it moves finaly to ICONS.

Static xyz.png saved vrom Preview Page do not work an need to be convertet to yxz.jpg

hope that helps.

Mayby only on fist flashed Upanzis ? (I have 2 with same results)

Blueforcer, 2023-10-25:
Sorry i cant reproduce it. 
It works for all other users without issues. I need to be something wich blocks the download. PiHole, any other blocker or somthing like this.

servicemensch, 2023-10-25:
Tested it by accessing from a different device (iPAd instead of Windows computer) to the display: Icon download successful. So the browser/device you are accessing the web interface matters in some cases. Anyhow all other icons were downloaded from the affected windows machine.

GM-Jokemaster, 2023-10-27:
Retryed it with Edge (Chromium) did do the trick (my Firefox doesn't)
Ty ! 
From my Point /closed

Glup3, 2023-12-05:
Hello, got the same issue.
Tried on MacOS with Firefox, Arc, Chrome & Brave.

Works on my iPhone tho - IOS 16

Blueforcer, 2024-02-21:
> Please reopen this issue ![image](https://private-user-images.githubusercontent.com/1973755/306579646-2ef2ae7e-e1db-46ab-9a67-478dc7e22a8e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MTAxNjQsIm5iZiI6MTcwODUwOTg2NCwicGF0aCI6Ii8xOTczNzU1LzMwNjU3OTY0Ni0yZWYyYWU3ZS1lMWRiLTQ2YWItOWE2Ny00NzhkYzdlMjJhOGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjFUMTAwNDI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzczY2JhYjFiOGYyOGM5OGNkYmRlMjY0YWZjODE0ODM1M2FiZGRkZWI0NDQ4ZWZmYzg4ZjVmYmZkYjllYmY4ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ._OZVL1k7jx4iKQMSz_qxAT9v0FTR-FxxMsDdR4S0_SM) There is problem with corsproxy.io. Maybe it's possible not to use this proxy and try to load icons directly from lametric?

Thats not possible, because your browser downloads the icon, not the ESP itself. Im using it as workaround to not getting a CORS violation. dont know why youre getting one..

Anonym-tsk, 2024-02-21:
> Thats not possible

Just use fetch with {mode: 'no-cors'}, it works.
I can't find source of awtrix web, only compiled js.

Anonym-tsk, 2024-02-21:
[image]
I don't know the reason, but my ip is blocked on corsproxy. According to this issue, i'm not alone.

Blueforcer, 2024-02-21:
> [image]
> I don't know the reason, but my ip is blocked on corsproxy. According to this issue, i'm not alone.

Not really. The root issue from the other users was the browser not corsproxy, maybe a browser extension.
You have a different issue, wich  of course I unfortunately unable to solve.

Anonym-tsk, 2024-02-21:
So, i changed function `downloadLametricImage` in `DoNotTouch.json` and now it works without proxy. Please try this solution.
[image]

Anonym-tsk, 2024-02-21:
@Blueforcer what do you think?

Anonym-tsk, 2024-02-29:
@Blueforcer Have you tested my solution? Maybe we can add it to the firmware and get rid of the proxy?

Blueforcer, 2024-02-29:
Not yet, im not at home this week.

But awtrix doesn't contain a Proxy . A Proxy usually sits in front of the awtrix to access it via https from the outside.

Anonym-tsk, 2024-02-29:
> But awtrix doesn't contain a Proxy . A Proxy usually sits in front of the awtrix to access it via https from the outside.

Yes I understand this.
But using an external proxy is not very secure (we can’t trust its creators 100%) and is unstable (it doesn’t work for me).

rusnasonov, 2024-03-07:
> I don't know the reason, but my ip is blocked on corsproxy. According to this issue, i'm not alone.

Just use VPN

adrienrichmond, 2024-04-22:
Same problem, do you have a fix ? 

[image]

Blueforcer, 2024-04-22:
Try this
[firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)

adrienrichmond, 2024-04-23:
> Try this [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)

Hey, not working

pacroy9, 2024-04-23:
> > Try this [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)
> 
> Hey, not working

I have the same problem, firmware update is unsuccessful

Blueforcer, 2024-04-23:
@Anonym-tsk I implemented your changes. Please check

Anonym-tsk, 2024-04-23:
> @Anonym-tsk I implemented your changes. Please check

Where to get the binary?

Blueforcer, 2024-04-23:
> > @Anonym-tsk I implemented your changes. Please check
> 
> Where to get the binary?

3 posts above

Anonym-tsk, 2024-04-23:
> I implemented your changes. Please check

How to do clean install. Now I have modified json. What I need to do? Reset before flash?

Blueforcer, 2024-04-23:
Just upload the bin file in the update tab of awtrix. 
Maybe it's necessary to delete the DoNotTouch.json and reboot awtrix. But not sure right now.

adrienrichmond, 2024-04-23:
Hey Blueforcer, do you have a solution to modify bin files ? thx!

Blueforcer, 2024-04-23:
Why you need to modify bin files? Your fix is already in the posted file

adrienrichmond, 2024-04-23:
Hello, j'ai flasher le firmware bin mais ceci ne fonctionne toujours pas merci!

Anonym-tsk, 2024-04-24:
> Try this
> [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)

It works, but (!):
You need to remove file `DoNotTouch.json` from filesystem after update and restart awtrix.
Without removing i have old file and the fix not applyed.

Anonym-tsk, 2024-04-24:
@pacroy9 @adrienrichmond try [this](#issuecomment-2074366565) please

adrienrichmond, 2024-04-24:
> > Try this
> > [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)
> 
> It works, but (!): You need remove file `DoNotTouch.json` from filesystem after update and restart awtrix. Without removing i have old file and the fix isn't applyed.

It's working, thank you very much bro :)

Anonym-tsk, 2024-04-25:
> Try this
> [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)

Something is wrong with this firmware.  I get unexpected device reboots from time to time.

pacroy9, 2024-04-25:
> @pacroy9 @adrienrichmond try [this](#issuecomment-2074366565) please

This firmware update and the deletion of the file solved the problem. Many thanks for the fix.

Schlicki, 2024-04-29:
Does not work for me. Got the modified firmware and deleted the "do not touch" but still get the error when trying to download any ID.

xbmcnut, 2024-05-02:
Just got my first TC001 and love it but have the same issue. Device is running 0.96 and I keep getting the `This ID doesnt exist` error when trying to download icons. Tried the firmware file listed in the various threads and deleted the `/DoNotTouch.json` file but then I lost access to the device on my network even though the device told me it had a network IP address (and the AP was off). Re-flashed to get it back online but now stuck without icons again. Tried FF, Edge and Chrome.

rtrive, 2024-05-04:
> > Try this
> > [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)
> 
> It works, but (!): You need remove file `DoNotTouch.json` from filesystem after update and restart awtrix. Without removing i have old file and the fix isn't applyed.

Confirm it works. Thanks

tebald1, 2024-05-15:
> Try this [firmware.zip](https://github.com/Blueforcer/awtrix3/files/15065354/firmware.zip)

Worked for me thx

serfin64, 2024-05-16:
Manually transferred the file with the icon to the Icons folder
[image]

I call icons to appear on the watch - the watch reboots

[image]

Displaying Effects and messages on the watch - works

Blueforcer, 2024-05-16:
i dont know this app, this is not the official app.
try to call it via API and make sure it 8x8

serfin64, 2024-05-17:
Icon - *.gif.
The clock is displayed as an animation in the web interface

https://github.com/Blueforcer/awtrix3/assets/156530596/5827b8aa-f197-4bc0-affc-75cdc8c14c61

Blueforcer, 2024-05-17:
A webpage can render everything, it has nothing to to with the matrix renderer.
But again, its not an 8x8 icon.
8x8 Icons doesnt have a black border nor a grid
it looks like this
[image]

rolfvreijdenberger, 2024-07-30:
adding some information here: I have the same issue, but this is caused by my pfsense distribution using pfblockerng (domain blacklist). Whitelisting corsproxy.io solves the issue.
Keep in mind, that this domain corsproxy.io is being flagged as malware by opensource blocklists. 
[image]

carltonwb, 2024-09-22:
Pardon my ignorance. I am a new user and am running into the same or at least similar problem.
Can someone please explain how to use the firmware.zip file and also where I will find the DoNotTouch.json file as well.
Thanks

captn3m0, 2024-11-20:
fwiw, corsproxy.io is entirely blocked within India (by corsproxy themselves).

```
curl -i "https://corsproxy.io/?https://developer.lametric.com/content/apps/icon_thumbs/15974"
HTTP/2 403
date: Wed, 20 Nov 2024 19:10:03 GMT
content-type: application/json
content-length: 123
report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=ssAX%2BqPRdfXHnyQUZwDdtuHbwd5kpvNhGWsXCsQsoZ1qUjtxSc30fJ1YN7cqasmyuFM%2Fm4JLOSw7WmwN7WunABxDg3fd%2BQ%2BGhiRNHBxsZmZEZ8dr%2FyU5LelqxJAkGQ%3D%3D"}],"group":"cf-nel","max_age":604800}
nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
strict-transport-security: max-age=2592000; includeSubDomains; preload
x-content-type-options: nosniff
server: cloudflare
cf-ray: 8e5abce69d538025-MAA
server-timing: cfL4;desc="?proto=TCP&rtt=8131&sent=5&recv=7&lost=0&retrans=0&sent_bytes=3395&recv_bytes=767&delivery_rate=268481&cwnd=251&unsent_bytes=0&cid=1ba05238f4a6158f&ts=22&x=0"

{
  "error":{
    "code":403,
    "message":"Country blocked! Your country is blocked from accessing this ressource!"
  }
}
```

While the preview works, the download doesn't. Would be nice if the corsproxy was configurable.

Blueforcer, 2024-11-20:
> fwiw, corsproxy.io is entirely blocked within India (by corsproxy themselves).

Please try the firmware some posts above.

tobydarkh, 2024-12-03:
> > fwiw, corsproxy.io is entirely blocked within India (by corsproxy themselves).
> 
> Please try the firmware some posts above.

Have the same error but with the firmware from the post...

Blueforcer, 2024-12-03:
> > > fwiw, corsproxy.io is entirely blocked within India (by corsproxy themselves).
> > 
> > Please try the firmware some posts above.
> 
> Have the same error but with the firmware from the post...

clear your browsers cache, otherwise have a look at the browsers developer console log

tobydarkh, 2024-12-03:
> > > > fwiw, corsproxy.io is entirely blocked within India (by corsproxy themselves).
> > > 
> > > 
> > > Please try the firmware some posts above.
> > 
> > 
> > Have the same error but with the firmware from the post...
> 
> clear your browsers cache, otherwise have a look at the browsers developer console log

Have clear the cache, but same erro: 
https://images.tobydarkh.com/2024/12/chrome_Tz9kdVquH4.png

https://images.tobydarkh.com/2024/12/chrome_9VqHQCep5Y.png

Blueforcer, 2024-12-03:
That's not the firmware I mentioned in this thread. This doesn't use corsproxy.

Elitehoernchen, 2024-12-17:
test firmware is working for me with that fix :)

Blueforcer, 2024-12-17:
Thx for the feedback. I will release the firmware the next days

snoopy12345, 2024-12-21:
Firmware solves the icon issue, but the Ulanzi reboots randomly from time to time. Something is wrong here.