Blueforcer/awtrix3 #366: [FEATURE REQUEST] Clock more color modes pls

drtekGER, 2023-10-26:
# Feature Request

### Is your feature request related to a problem?

If so, then add a clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

### Describe the solution / feature you'd like

Clock  App needs more color and blinking separators please. Rainbow mode and or color changing mode for clock would be super.
I am using TC001 just as a basic clock since I am a basic user. 
Bought the app and will consider donating if my wish will be heard.
Thanks!

### Describe alternatives you've considered

A clear and concise description of any alternative solutions or features you've considered.

### Additional context

Add any other context or screenshots about the feature request here.

Blueforcer, 2023-10-26:
You're already able to change the Text color and add blinking to the separators. You can also change the color of the weekday bars

drtekGER, 2023-10-26:
Yes I know and enjoy but just a little more color fading / rainbow animation like the one during startup would make this clock a real eyecatcher.

Gotta look into how to make the : blink, thanks for fast reply

Blueforcer, 2023-10-26:
%H %M %S doesnt exist. because it already shows the seconds.
The blinking colon is only for formats wich doesnt show the seconds.

drtekGER, 2023-10-26:
Okay thanks, please consider adding mentioned color options to bring some life into my office. Danke aus Schweinfurt

ThePituLegend, 2023-11-01:
I would really love to get back the "global" rainbow mode (as it was in Awtrix 2.0). I really loved that. I can use rainbow in flows, but since date and time are not flows, I can't (or I don't know how to) change it...

receiverbox, 2023-12-28:
Actually, I expierinced some issues iwth 0.93 version too regarding the clock. 
As the calendar Icon is shown by default, I cannot get %H %M %S displayed.
Indeed I think there should be the option to have also the FULL HEIGHTS of the display be available for clock mode optionally. Today it only uses 5 pixel rows out of 8, so the numbers would be really much bigger with using 8 rows.

Blueforcer, 2023-12-28:
> Actually, I expierinced some issues iwth 0.93 version too regarding the clock. As the calendar Icon is shown by default, I cannot get %H %M %S displayed. Indeed I think there should be the option to [...]

The AWTRIX Font is based on TomThumb 3x5 wich doesnt allow to resize the font. We would need a complete new one.
However, using the entire height would not help at all in this case, on the contrary, you would need even more in the width. If you want to show also seconds, you can turn off the calendar icon to have the full 32 pixel available (see setting TMODE)

receiverbox, 2023-12-31:
Actually this is two different observations: I noticed that %H %M %S by default is obviously without %S - havent yet tested switch calender Icon off). 

The other aspect is to have the time in max size (%H %M), this requires then only 0 - 9 as new font icons of course. Not sure if people are intressted in running test with 4x8 Letters.

closed 2024-06-11 by Blueforcer