Blueforcer/awtrix3 #369: [BUG] Question about UTF-8 special character support

alfureu, 2023-10-28:
# Bug report
This is not necessarily a bug report, but I was trying to put UTF-8 special characters to display with HomeAssistant to Ulanzi using this ROM, and before I dig deeper, I would love to know whether the font-set available supports characters like these: ↗↘→⇈⇊

I know there are the corners that can be triggered, but they are insufficient as you can see from the characters I would like to use

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.90

Blueforcer, 2023-10-28:
These characters are not supported and it would be a bit tricky to show them in 8 Pixel height.

You can find all supported characters in the font file
https://github.com/Blueforcer/awtrix-light/blob/main/src/AwtrixFont.h

alfureu, 2023-10-28:
Thanks, so the only option is to introduce a variety of icons...

Blueforcer, 2023-10-28:
Yes, or drawing commands

alfureu, 2023-10-28:
can you elaborate pls? I am unaware of drawing commands...

Blueforcer, 2023-10-28:
https://blueforcer.github.io/awtrix-light/#/api?id=drawing-instructions

alfureu, 2023-10-28:
Thanks, closing this ticket then

closed 2023-10-28 by alfureu