Blueforcer/awtrix3 #374: [BUG] - Looks like Coloring of fragmented text wont work & Color defintion is inconsistent

Bitfummler, 2023-10-29:
Hi, I really enjoy your work.

Just discovered an issue with fragmented text. The color defintion won't be applied properly, Color value "4000FF" results into purple colored text fragment. Any color value "5000FF", "6000FF", ... "F000FF" results into solid blue colored text fragment without any variation in color. Here I woud expect that there are multiple variations of purple. 

I als want to point out that in the doc all color setting in hex notation are specified as "#rrggbb" while for text fragments it is specified as "rrggbb" without the leading hash char.

Blueforcer, 2023-11-12:
Im unable to reporduce it.

```
{
  "text": [
    {
      "t": "1",
      "c": "0000ff"
    },
    {
      "t": "2",
      "c": "4000FF"
    },
     {
      "t": "3",
      "c": "6000FF"
    },
     {
      "t": "4",
      "c": "8000FF"
    },
     {
      "t": "5",
      "c": "F000FF"
    }
  ]
}
```

gives me

[image]

it doesnt matter if youre using hash char or not.

Bitfummler, 2023-11-12:
I used this test json:

{
  "textCase": 2,
  "scrollSpeed": 100,
  "repeat": 1,
  "text": [
    {
      "t": "1",
      "c": "3000FF"
    },
    {
      "t": "2",
      "c": "4000FF"
    },
    {
      "t": "3",
      "c": "5000FF"
    },
    {
      "t": "4",
      "c": "6000FF"
    },
    {
      "t": "5",
      "c": "7000FF"
    }
  ],
  "textOffset": 0,
  "icon": 13593,
  "pushIcon": 0,
  "center": "true"
}

When I capture the output via the http interface I get this:

[image]

You can see the color variations, but the watch shows me something else. So, I assume, the read back display data doesn't refelct the reality.

I figued out it has to do with brightness. I set min brightness level to 1 and then I get 1st 2 chars in same blue und last 3 one in same purple color. With BRI=2 I get all chars in same purple, BRI=3 seems to be the 1st level which works.

Blueforcer, 2023-11-12:
The colors are off on very low brightness since Beginning.
Because the LEDs doesn't get enough power.
No fix found so far.

But has nothing to to the fragmented texts.

closed 2023-11-12 by Blueforcer

Bitfummler, 2023-11-12:
Ok, my learning curve. Let's close this ticket.

Blueforcer, 2023-11-12:
If you find a way to solve this problem, I'm happy to discuss this with you

Bitfummler, 2023-11-12:
Need to think about it, but are limited in time right now ;) I guess, it needs a calibration procedue controlled by the user since it might be a HW specific property. Goal should be to determine for any base color the min/max value of each base color. If this is known, the calibration data can be used to scale the color settings. Just a guess so far, it needs some experiments, if this really makes a diference.

I'm new here on Github, have other questions for you regarding development/contribution. How can I do that outside of a topic.