Blueforcer/awtrix3 #378: [FEATURE REQUEST] no auto change for "and" and "or" text in message field

technarf, 2023-10-31:
# Feature Request

In a custom app, when I send a message which contains the words "and" or "or" (I'm french so the real words are "et" and "ou", but that is the same in english), they are automatically changed in "&&" and "||" like if they were in a test condition. As my text is a phrase, it is really annoying to have this automatic change. If I write two "and" or two "or" in a row, the first one is converted, not the second one. I can't find a way to disable this "feature", or behavior... 
So, I'd like to have my message field not to be transformed from the original text. Is there a way to do that or not ?
Thanks for your help, and sorry if a solution already exists...

Blueforcer, 2023-10-31:
There is no automatic replacement of any words.
Please check the raw JSON wich is sent to awtrix. 
 I'm pretty sure its your server wich sends this

technarf, 2023-10-31:
My bad, you're totally right, I've just discovered in mqtt explorer that the message sent was modified. I have to investigate but thank you for your help...

closed 2023-10-31 by technarf