Blueforcer/awtrix3 #383: [FEATURE REQUEST] - Need opton to hide a customer app

Bitfummler, 2023-11-01:
I understand the concept of deleting customer app, but this has a drawback. When I add this app again, it will be added at the end of the app list. 

I'm looking for a way to hide apps via API in order to ensure a static transition list with the option to skip displaying apps when their content might be not from interest (e.g. when a PV system doesn't deliver energy).

Any attempt to solve this issue via existing API and an MQTT controller for the clock doesn't look nice from visual perspective, since the only way of achieving my gool seems to be that I need to mange the list of apps in the MQTT controller such that any change of that list leads to deleting all apps before I pushed the new list completely. This will destroy the onging automated transition sequence and the clock starts to display with its 1st internal app which is actice.

Blueforcer, 2023-11-01:
The current scaffold doesnt allow this, and it would be a huge effort for this single request.
You have 3 options:
- Control the loop for your own with this https://github.com/Blueforcer/awtrix-light/issues/381
- You can use the "pos" json key to set the index in the loop (highly experimental)
- Send a json list of all app names to /api/reorder. it should work if you delete and add the name to the list

Bitfummler, 2023-11-01:
Thanx for this hint, will test that later and provide feedback.

closed 2023-11-28 by Blueforcer