Blueforcer/awtrix3 #387: [BUG] Battery charge at 100%

FeromonDE, 2023-11-04:
# Bug report

## Describe the bug

If the battery charge reaches 100%, the charge changes almost every second in HA. 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.90 ]

## To Reproduce

Charge to 100%

## Expected behavior

See the battery gradually discharge

## Screenshots

[image]

Blueforcer, 2023-11-04:
Im not the developer of the hardware 😜
It's not a firmware bug

closed 2023-11-04 by Blueforcer

FeromonDE, 2023-11-04:
It turns out that the battery charge remains unknown and the device can turn off at any moment?

Blueforcer, 2023-11-04:
> It turns out that the battery charge remains unknown and the device can turn off at any moment?

Im not sure what you mean.

Kannst auch gern auf deutsch schreiben.

FeromonDE, 2023-11-04:
I just wanted to monitor the battery charge and at a certain value turn on charging through the smart socket and turn it off at 100%

Blueforcer, 2023-11-04:
I don't see any problem to archive this.
If percentage > 95% longer than 1min, turn off.

But the cheap battery will die pretty fast if you always push full charging cycles.

FeromonDE, 2023-11-04:
> I don't see any problem to archive this. If percentage > 95% longer than 1min, turn off.
> 
> But the cheap battery will die pretty fast if you always push full charging cycles.

[image]
Ladestand innerhalb von 1.5 Minuten, und so springt die Ladung hier und her fast jede 5 Sekunden.

Blueforcer, 2023-11-04:
Dann baue eine Mittelung ein wenn dir das zu viel ist und gehe auf 2min Verzögerung.
Die Werte kommen so von der Hardware und werden bereits von der Software geglättet.
Die Batterie und Hardware an sich ist halt Schmutz.