Blueforcer/awtrix3 #389: [BUG] Selecting DoNotTouch.json and clicking MkFile erases the file

jankazemier, 2023-11-07:
1. I clicked on the DoNotTouch.json file (not even wanting to edit it)
2. clicking the file entered the filename into the box 
3. clicking MkFile immediately overwrites the file with an empty file
4. DoNotTouch.json is overwritten without any warning or way to discard changes or to get it back

[image]

jankazemier, 2023-11-07:
Ok, I managed to restore it by saving the config again using the webinterface. Still, this should not happen :)

Blueforcer, 2023-11-07:
It's the behavior of the ACE file editor.
It's not made by me, you should open an issue there.

If you delete the JSON, awtrix will generate a new one after reboot.

closed 2023-11-07 by Blueforcer