Blueforcer/awtrix3 #390: [BUG] MQTT prefix is used as client ID

ahk97, 2023-11-09:
# Bug report

## Describe the bug

Awtrix light uses the configured "MQTT prefix" as the client ID when connecting to the MQTT broker.

This prevents two Awtrix devices with the same prefix connecting to the same broker

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.90

## To Reproduce

Steps to reproduce the behavior:
Configure MQTT and check mosquitto log.

```
1699536844: New connection from 192.168.178.16:60894 on port 1883.
1699536844: Client awtrix/1 already connected, closing old connection.
1699536844: New client connected from 192.168.178.16:60894 as awtrix/1 (p2, c1, k15, u'awtrix2').
1699536844: New connection from 192.168.178.15:57297 on port 1883.
1699536844: Client awtrix/1 already connected, closing old connection.
1699536844: New client connected from 192.168.178.15:57297 as awtrix/1 (p2, c1, k15, u'awtrix1').
```

## Expected behavior

The client should choose a unique ID (for example based on the MAC Address  or hostname) or the Client ID should be configurable.

## Additional context

It is necesessary to use the same prefix on two Displays to have them display the same content

Blueforcer, 2023-11-09:
That's not a bug.
It rises historical.
And will be used in lots of functions.

Standard Prefix/Client is a unique ID if you don't change it.

This behavior will not be changed because now hundreds of users would need to change their systems.

Just use different prefix to all your awtrix. It's also easier for you to differ the clocks in your automations.

closed 2023-11-09 by Blueforcer

Blueforcer, 2023-11-09:
Let me think about it this afternoon.
The Webinterface is shit and every change to it would force all users to reset their config.

ahk97, 2023-11-10:
I checked the MQTT documentation and it says the client identifier must be unique:
https://www.ibm.com/docs/en/ibm-mq/9.3?topic=concepts-client-identifier

Also, just changing the client ID in the code to a unique value should not break any behavior.
As long as the standard prefix (MQTT subscription topic) stays the same the functionality would stay the same as well.

Thanks!

Blueforcer, 2023-11-10:
I know how mqtt works, but since today you're the first out of thousands users with this "issue".

HA discovery also uses the client ID.
But as said before I need to check it if this could break something for existing users.