Blueforcer/awtrix3 #391: Battery level reporting may be wrong

luebbe, 2023-11-09:
# Bug report

## Describe the bug

Maybe a bug, maybe a question...

My Ulanzi Clock is powered by an USB hub while my PC is running. At night I turn off the clock's matrix via home assistant so it runs off the battery.
In the morning the clock is sometimes still running and the matrix can be turned on via home assistant. But most of the time it has shut itself down over the night. The attached screenshot shows the typical uptime vs. battery level of the Ulanzi recorded by home assistant over the course of a week.

[image]

You can see that the clock shuts itself down, when the reported battery level drops somewhere around 55%-60%. In the morning when I turn it on again, after some initial confusion (0%-100% transition) it starts with a reported battery level of 70%-80%.

Now my question: If the battery cannot keep the clock running anymore, shouldn't the reported battery level be close to 0% instead of 60%? To me this looks like the computation of the level is wrong.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)

## To Reproduce

Turn the matrix off and record the uptime and battery level.

## Expected behavior

The battery level should oscillate between 0% and 100%. The reported battery level should be near 0% when the shutdown occurs.

## Additional context

The zero to max transition of the battery level upon boot may be the same issue as #354

Blueforcer, 2023-11-09:
It seems that there are different batches out there where battery measurement works different. But it's impossible to detect this.
You can set a new min/max range via dev.json

closed 2023-11-28 by Blueforcer

luebbe, 2023-11-29:
I was thinking about implementing a little helper for the battery level calibration. If we collect the mit/max values of batt_raw over time (maybe better with some filtering/averaging), we could report this interval via mqtt and this would in turn allow us to set the min/max values in the dev.json file accordingly.

eku, 2024-03-17:
I was wondering how 8646eb4 is related to battery level?

luebbe, 2024-03-17:
looking at that changeset probably nothing at all :D

fragolinux, 2024-04-21:
> I was thinking about implementing a little helper for the battery level calibration. If we collect the mit/max values of batt_raw over time (maybe better with some filtering/averaging), we could [...]

hi, so, how to get the min easily? today mine died when battery was at 40% or similar, after 14 hours with automatic luminosity...

now i recharged it, i get this, how can i retrieve the bat_raw of when it dies? To reconnect i have to power it again, so read will be affected a bit and reading not accurate... when i connected it, after it settled a bit, after 1 minute or so i saw battery at about 80% on display, which of course could not be...

```
{
  "bat": 100,
  "bat_raw": 704,
  "type": 0,
  "lux": 0,
  "ldr_raw": 235,
  "ram": 135312,
  "bri": 3,
  "temp": 22,
  "hum": 34,
  "uptime": 18127,
  "wifi_signal": -48,
  "messages": 2,
  "version": "0.96",
  "indicator1": false,
  "indicator2": false,
  "indicator3": false,
  "app": "Humidity",
  "uid": "awtrix_6dbbe4",
  "matrix": true,
  "ip_address": "192.168.1.107"
}
```

luebbe, 2024-04-22:
You can use mqtt explorer or the home assistant integration to monitor the battery raw value over time. Then you will know which raw values correspond to 0% and 100% and add (values close to) these to your json config file.

fragolinux, 2024-04-22:
> You can use mqtt explorer or the home assistant integration to monitor the battery raw value over time. Then you will know which raw values correspond to 0% and 100% and add (values close to) these to your json config file.

thanks, didn't note that mqtt exposed that value, sorry, but now i'm already monitoring with a crontab with this line:

`* * * * * curl -s http://192.168.1.107/api/stats -o /nvme/Media/$(date +\%Y-\%m-\%d_\%Hh\%Mm).json`

as soon as it dies, i'll have my values :)

fragolinux, 2024-04-22:
done, it lasted about 13 hours with automatic brightness, do you think it's good?

this is when i disconnected the cable after fully charged:

```cat 2024-04-22_01h04m.json
{"bat":100,"bat_raw":709,"type":0,"lux":3,"ldr_raw":406,"ram":135148,"bri":11,"temp":23,"hum":34,"uptime":50306,"wifi_signal":-53,"messages":64,"version":"0.96","indicator1":false,"indicator2":false,"indicator3":false,"app":"Temperature","uid":"awtrix_6dbbe4","matrix":true,"ip_address":"192.168.1.107"}```

and this is the last file written by the cronjob polling stats every minute:

```cat 2024-04-22_13h49m.json
{"bat":57,"bat_raw":584,"type":0,"lux":1,"ldr_raw":319,"ram":134592,"bri":6,"temp":23,"hum":31,"uptime":96207,"wifi_signal":-82,"messages":64,"version":"0.96","indicator1":false,"indicator2":false,"indicator3":false,"app":"Time","uid":"awtrix_6dbbe4","matrix":true,"ip_address":"192.168.1.107"}```

isn't the original 57% a bit "too much" off? Or these retrieved values are normal? Thanks!

luebbe, 2024-04-23:
13 hours are very good and I can confirm your min/max values.
Mine ran for about ten hours from 18:00-4:15 in the morning this night. Sometimes it survives the night. I have a HA automation that turns the matrix off at midnight. Auto brightness as well.
Looks like we have batteries from the same batch in our awtrix. I set min_battery in my dev.json to 600, but I think I can go lower, because my awtrix continues to run for another 45 minutes, when the battery level shows 0%. max_battery is unchanged.
I think the inital values are not taken out of thin air, but are/were ok for previous batches of awtrix.

fragolinux, 2024-04-23:
thanks, very good job!

Moonbase59, 2025-03-17:
Just some feedback for others to find later:

Seems the Ulanzi clocks vary quite a bit regarding battery status reported. I bought mine in late summer 2024, and it usually runs on USB power. But since it _can_ run 5–12 hours off the battery, depending on room lighting and display on/off state, I sometimes take it with me into another room. (I have auto brightness on, and the display gets turned off when I go to bed, and turned back on at sunrise.)

Using the [current raw battery defaults of Awtrix 3 v0.98](https://blueforcer.github.io/awtrix3/#/dev?id=json-properties) (475/665), it typically shuts down at around 45% of shown battery percentage in Home Assistant.

So I added a simple raw battery sensor to Home Assistant, did a few full discharge/charge cycles, and ended up with these values for my `dev.json` file:

```json
{
  "min_battery": 560,
  "max_battery": 671
}
```

It now shows a more realistic battery percentage, and is more in line with my battery management automations (notify below 20%, including a message on the display itself).

pmacca, 2026-01-04:
For anyone who stumbles across this later, I have been able to easily expose the "bat_raw" to HA by adding this config to `configuration.yml`

```
mqtt:
  sensor:
    - name: "Awtrix Raw Battery"
      unique_id: "awtrix_05b994_bat_raw"
      state_topic: "smartclock/stats" # where "smartclock" is my MQTT prefix in the Awtrix MQTT config page
      value_template: "{{ value_json.bat_raw }}"
      unit_of_measurement: "mV"
      device_class: voltage
      state_class: measurement
      icon: "mdi:battery-high"
```

This has exposed the raw battery metric to HA. I plan to monitor the max/min value through a charge cycle and use it to callibrate the settings.

commit 8646eb4a (2023-11-28): v0.91

- Allows to set a http callback for button presses via dev.json.
- You can send a 8x8 jpg icon as base64 string.
-  Applies gamma correction to brightness control
- IP adress was added to stats
- Fixes a bug where duration wasnt reset after custom app deletion. closes #407
- Indicator fading and blinking options are now disabled after deactivation. closes #402
- Awtrix sends the stats only after the first full sensor reading to avoid zero numbers after boot. closes #391
- Uptime reading now handles the millis() overflow to calculate longer uptime than 49days
- fixes some minor bugs