Blueforcer/awtrix3 #392: [BUG] Timezone to another location

shawndamon, 2023-11-11:
# Bug report

## Describe the bug

I cant seem to update the timezone the following is what i have tried:
Asia/Shanghai,CST-8
Asia/Singapore,<+08>-8

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## To Reproduce

[Steps to reproduce the behavior:]

(http://awtrix_efbac8.local/) under time trying to change it

[image]

Blueforcer, 2023-11-12:
There is a link right here where you can find the setting you need:
[image]

So for Shanghai it is **CST-8** and it works

[image]

closed 2023-11-12 by Blueforcer

shawndamon, 2023-11-27:
sorry i didnt see you message on Discord sooner...thank you for closing this