Blueforcer/awtrix3 #394: How to use the drawing db command?

salabim, 2023-11-11:
I am trying to write an RGB888 bitmap array via HTTP to my awtrix device.
The documentation just says
`{"db": [x, y, w, h, [bmp]]}`
, but I can nowhere find the format for the bitmap bmp.
Just an example might already help.

BTW, all the other drawing instructions work for me!

salabim, 2023-11-12:
I've managed to work out that bmp is a string of w * h integers, composed of rgb value, like (in Python) for red
0xff0000.
Now can I show whatever I want!

closed 2023-11-12 by salabim

bmorg, 2024-10-22:
Could you provide an example? @salabim

salabim, 2024-10-22:
It was a question from my side.

bmorg, 2024-10-22:
Your second comment sounded like you had solved it yourself. :)

In any case, one of the LLMs knew the correct syntax:

```json
{
    "draw": [
        {
            "db": [
                0,
                0,
                3,
                3,
                [
                    16711680,
                    16711680,
                    16711680,
                    65280,
                    0,
                    65280,
                    255,
                    16777215,
                    255
                ]
            ]
        }
    ]
}
```

The color values are just the RGB values as integers, e.g. `0x00ff00` is  `65280`.

salabim, 2024-10-22:
Sorry for my confusing answer.

This code fragment shows how I solved it:

    def show_image(self, image, x=0, y=0, name="image"):
        image_as_bytes = image.tobytes()
        image_as_array = [(b0 * 0x10000 + b1 * 0x100 + b2) for b0, b1, b2 in zip(image_as_bytes[0::3], image_as_bytes[1::3], image_as_bytes[2::3])]
        self.custom(name=name, draw=[{"db": [x, y, *image.size, image_as_array]}], duration=1)