Blueforcer/awtrix3 #395: Make draw apps scrolling

salabim, 2023-11-12:
It would be nice if we could have draw apps, which are wider than 32 pixels could scroll, like other apps.
This is particularly useful for pictures specified as bitmaps.

Blueforcer, 2023-11-12:
What is a draw app?

salabim, 2023-11-12:
I mean a custom app which has a key 'draw'.
And for me, the drawing instruction 'db' therein is most interesting.

Blueforcer, 2023-11-12:
Drawing commands are always static. If you want to move something you need to change the coordinates. 
There will be no way to do this automatically

salabim, 2023-11-12:
That's a pity.
My use case is a 'long' text, made from another font, that I would like to scroll like a text app. But, ow I can't, as far as I know.
Could you add to the feature request list?

Blueforcer, 2023-11-12:
It's not that trivial as it sounds.
Awtrix can't count the pixels of each characters if it's just a picture. This is necessary to scroll the text

salabim, 2023-11-12:
My characters are just an example. 'All' I ask is to scroll the picture one pixel at a time, really like the text app.
But maybe I don't understand the system enough. Anyway, I love awtrix!

Blueforcer, 2023-11-12:
Just update the position of the picture. Awtrix has no problems in receiving fast app parameter changes.

There will be no Automation in the drawing functions, sorry.

closed 2023-11-12 by Blueforcer

salabim, 2023-11-12:
Ok.
Thanks.