Blueforcer/awtrix3 #396: Device turns off after roughly 10min

Kefut, 2023-11-13:
The clock was running without issues until the weekend, where it started to just turn off. I can turn it back on by pressing the to arrow buttons, but it will turn off after a few minutes (roughly 10min)  

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.9

Steps to reproduce the behavior:  I turn it on and wait for a few minutes (roughly 10min)  

## Logs

I cannot find the reason for this sudden behaviour and I don't see an option for a log.

Blueforcer, 2023-11-13:
The firmware has nothing to do with the Powermanagement, therefore it's not possible to turn the clock on or of from software side.
Mostly this happens because of a weak power supply. The batter discharge faster then it can load

Blueforcer, 2023-11-13:
You can connect it to USB and listen to the serial output.

But of you need to power it on with two buttons, it's really turned off on hardware side and so the ESP32. Nothing I can detect from software.

Blueforcer, 2023-11-13:
Nowadays there are clocks out there wich have different battery measurement. And it's off by 50%.
But there is no way to detect this programmatically.

So in your case I'm pretty sure the battery is empty. If the charging current is higher than the consumption it should raise. Try using a 5V with at least 2A.

Kefut, 2023-11-13:
I checked with a different power supply and it's still on after 45min. Thanks for you super fast feedback.

Kefut, 2023-11-13:
Not a software problem

closed 2023-11-13 by Kefut