Blueforcer/awtrix3 #399: Nach statischer IP Vergabe bekomme ich keine Uhrzeit und kein Datum mehr vom Zeitserver

schorchotto, 2023-11-14:
Ich weiß nicht ob es ein Bug ist oder ich zu blöd.
Ich habe 2 TC001, beide mit Awtrix light geflasht und mit einer statischen IP versehen im Netzwerk. Funktioniert soweit auch bis auf Uhrzeit und Datum die werden nicht mehr vom Zeit Server eingeholt.
Awtrix Firmware 0.90 (hatte das Problem aber auch schon bei 0.78)
Habe in der Fritzbox den Statischen IP Bereich vergrößert.
[image]
[image]
[image]
Gruß
Christoph

Blueforcer, 2023-11-16:
Nutze ich und viele andere bereits seit der ersten Version.
Gab bis dato noch keine Probleme. Ich gehe davon aus das dein DNS den Zeit Server nicht auflösen kann oder ähnliches.

aufm Discord hatte jemand das selbe Problem. Es lag wohl an einer Einstellung in seinem Router.
Genauer hat er es nicht ausgeführt
https://discord.com/channels/546407049148366859/1164124015300968470

Blueforcer, 2023-11-28:
Neue Erkenntnisse zu diesem Problem?

tk1, 2023-12-04:
Ich habe dasselbe Problem. TC001 mit awtrix 0.91. Der log von pihole sagt:

2023-12-04 12:01:42 | A | de.pool.ntp.org | awtrix_5637b0.xxx,yyy | OK (sent to resolver1.opendns.com#53)

Datum und Uhrzeit werden aber nicht eingestellt. Die Uhr läuft um 1:00 los.

sysadmin-info, 2023-12-07:
Stupid question. Can you change the pool server in ulanzi tc001 and check? Can you see the communication on fritz to the time server? Can you connect directly from fritz to time server? Did you set proper time zone in ulanzi tc001 because it can be related to this (I guess).

tk1, 2023-12-07:
> Stupid question. 

Please stay polite. But thank you for your questions.

> Can you change the pool server in ulanzi tc001 and check?

same problem, i.e. if I boot the clock, it will contact the time server, get an answer (according to the log of my pihole), but the clock starts at 1:00.

> Can you see the communication on fritz to the time server? Can you connect directly from fritz to time server?

No fritz here. But when checking on my desktop computer, there is no problem contacting the time server.

```
w32tm /stripchart /computer:de.pool.ntp.org /dataonly /samples:5

de.pool.ntp.org wird verfolgt [178.215.228.24:123].
5 Proben werden gesammelt.
Es ist 07.12.2023 09:58:52.
09:58:52, +00.0793988s
09:58:54, +00.0818392s
09:58:56, +00.0826439s
09:58:58, +00.0834413s
09:59:00, +00.0792902s
```

> Did you set proper time zone in ulanzi tc001 because it can be related to this (I guess).

CET-1CEST,M3.5.0,M10.5.0/3

sysadmin-info, 2023-12-07:
I meant that my question is stupid not yours. Anyway I apologize for this.

sysadmin-info, 2023-12-07:
I am just wondering is the time zone set correctly just because in ulanzi
there is a link and if you will click it it will show you a csv file with
all the timezones you can set, so it should be set for your location where
you live. This is something that I can advise. Also can you check from
pihole machine connectivity to the time server? I am just curious. From
desktop is working so pihole does not block the time server. Anyway you can
try to change the time server to any other from Microsoft for example and
check it.

tk1, 2023-12-07:
Thank you for your suggestions.

The logs of pihole for awtrix and desktop (Falke) say:

```
2023-12-07 17:04:51 | A | de.pool.ntp.org | awtrix_5637b0.xxx.org | OK (answered by dns.opendns.com#53) | IP (43.4ms)
2023-12-07 17:13:22 | A | de.pool.ntp.org | Falke.xxx.org | OK (answered by dns.opendns.com#53) | IP (23.9ms)
```

Then I tried to specify the NTP server by IP: 88.99.86.9 and took the timezone EET.

But the problem remains, If I start the TC001 (awtrix 0.91) the clocks starts now at 0:00 because of the change timezone (was 1:00).

sysadmin-info, 2023-12-09:
Hi,

This is really strange behavior. I have the only two suggestions: to
upgrade Pi Hole to the newest version, but I doubt it is necessary, because
the traffic is allowed and Pi Hole does not block it.

To use instead of IP address some domain NTP server name. Look for time
servers and check just in case.

The last thing that I can eventually suggest it to reflash the Ulanzi TC001
using flasher on Blueforcer website. If the device is not recognized by the
flasher, the driver is needed. You will find it on my website. Check my
github profile . I created yesterday two articles how to flash the Ulanzi,
when it is not recognized by flasher.

tk1, 2023-12-11:
Hello, thank you once more. I tried your suggestions, but the problem remains.

I decided now to write a script which sends the time and date via mqtt to the clock.

AngieSoft, 2023-12-13:
versuch mal, die fb (fritz.box) als Zeitserver zu nutzen, das macht/kann die nämlich normalerweise.

SaPoX76, 2023-12-28:
Hast Du mal versucht die statische IP über die FritzBox bereitzustellen und nicht am TC001 zu konfigurieren? So funktioniert das zumindest bei mir.

tk1, 2024-01-02:
Die Tipps bzgl. FRITZ!Box haben leider nichts gebracht.

tk1, 2024-02-22:
Good news: after the installation of version 0.94, date and time are shown correctly.

Blueforcer, 2024-03-08:
@schorchotto Existiert das problem noch oder kann deises issue geschlossen werden?

closed 2024-03-17 by Blueforcer