Blueforcer/awtrix3 #400: [FEATURE REQUEST] Instructions to restore original firmware

Sumidu, 2023-11-16:
# Feature Request

### Is your feature request related to a problem?

No
### Describe the solution / feature you'd like

I would like to have a webflasher for the original firmware! 

### Describe alternatives you've considered

I have tried looking for clear instructions on the web, but they are all fuzzy.

### Additional context

After installing AWTrix Light, I found no way of how to go back to stock.

Blueforcer, 2023-11-16:
You can find the firmware on their website under "Tips"
https://www.ulanzi.com/products/ulanzi-pixel-smart-clock-2882

2sucres, 2023-11-20:
Not enough space when flashing with the web interface from the device

Sumidu, 2023-11-20:
I found the firmware but no instructions. 🫢

Dunkelschunkel, 2023-11-22:
I had the same problems. I think I was using a web flasher and only managed to get it up and running after flashing the original firmware twice. Even then it was not reboot safe.

Blueforcer, 2023-11-28:
[image]
[ulanzi_original_firmware.zip](https://github.com/Blueforcer/awtrix-light/files/13491066/ulanzi_original_firmware.zip)

closed 2023-11-28 by Blueforcer

Blueforcer, 2023-11-28:
> Sorry to bother again. Is there an equivalent MacOS tool somewhere? 

I don't know, sorry. But I'm pretty sure there are flash tools for Mac available.