Blueforcer/awtrix3 #401: ability to rotate the whole display

timgentschhh, 2023-11-16:
# Feature Request

It would be nice if I could rotate the display for 180 degree , so that I could stick it on the underside of my kitchen cupboard.

Blueforcer, 2023-11-16:
Already possible:
https://blueforcer.github.io/awtrix-light/#/dev

timgentschhh, 2023-11-16:
very nice! thank you very much for your quick reply
-----------------------

closed 2023-11-16 by Blueforcer

bachoo786, 2023-11-24:
@Blueforcer sorry I have just got my ulanzi display and I really need to rotate my screen. I know you have shared the link above but how do I implement it using JSON ? thanks.

Blueforcer, 2023-11-24:
> @Blueforcer sorry I have just got my ulanzi display and I really need to rotate my screen. I know you have shared the link above but how do I implement it using JSON ? thanks.

Create a dev.json in the Filemanager
And add
{"rotate_screen":true}

bachoo786, 2023-11-24:
> > @Blueforcer sorry I have just got my ulanzi display and I really need to rotate my screen. I know you have shared the link above but how do I implement it using JSON ? thanks.
> 
> Create a dev.json in the Filemanager And add {"rotate_screen":true}

hey thanks for the reply so I did what you told me to do and restarted the ULANZI display however it has not rotated the screen. Please see below:

[image]

Blueforcer, 2023-11-24:
The Filemanager tell you the error in your JSON. There is a leading space marked as a red dot. Remove it.

bachoo786, 2023-11-24:
> The Filemanager tell you the error in your JSON. There is a leading space marked as a red dot. Remove it.

thanks works perctly