Blueforcer/awtrix3 #404: Problem with icons in custom applications

luebbe, 2023-11-17:
# Bug report

## Describe the bug

While working on #354, I noticed that the current head revision has a problem displaying the icons of custom applications. They appear or disappear and the behaviour can be reproduced consistently.
It doesn't matter if auto transition is on or off. Manual transition has the same effect.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: main head revision

## To Reproduce

This is the observed behaviour:
- When the custom application is shown, the icon disappears after a moment (half a second, whatever) and doesn't appear again.
- When I manually toggle next and then previous app, the custom app is initially shown without icon, the icon appears after a moment and stays.
- A following next/prev toggle shows the app initially with icon again, which disappears after a moment.

## Screenshots

Attached is a gif which shows the behaviour. Manual next/prev toggle.

[image]

Blueforcer, 2023-11-17:
That's weird.
Mine running without such glitches and never heard about this from my beta testers.
Did you tried to turn the device off and on ? 🤪

luebbe, 2023-11-18:
> Did you tried to turn the device off and on ? 🤪
More than once :)

I'm running my version with the fix for #354, which you can find here: https://github.com/luebbe/awtrix-light. Since you know your code better than me, maybe you can spot a problem that I have missed. I wouldn't expect such a side effect. And why only for custom apps?

Blueforcer, 2023-11-18:
I I haven't looked at it yet.
But to be sure if you have built in the bug, you might want to test my main first.
At the moment I don't have the time to fix the bugs of others  😅

luebbe, 2023-11-18:
I'll test with your main withour my changes :)
I also noticed something else which is strange with all official versions from Awtrix since 0.84. Sometimes the bottom left pixel of the matrix turns yellow. This isn't reproducible, but when this happens, the web interface doesn't work anymore. MQTT ist still working. 
It is like an early indicator of an incoming severe failure.

Blueforcer, 2023-11-18:
This fading yellow pixel means the mqtt connection is lost. Top edge red pixel means lost wifi connection

luebbe, 2023-11-18:
Ok, so the yellow pixel is no accident. Haven't seen the red one yet. Is this documented anywhere?

I have the same graphic glitches with the icons in custom apps using your main without my changes. The gif would look the same, so I didn't shoot a new one. So this bug must have been introduced somewhere since the 0.90 release.

Blueforcer, 2023-11-18:
So there must be something in your config, automation or anything else special wich cause this.
Since today I never heard about such effect, and I'm also unable to reproduce it.

luebbe, 2023-11-18:
I just flashed the official 0.90 binary and the problem is gone. So it's definitely a problem in the head revision.

Blueforcer, 2023-11-18:
Good to know, but weird it doesn't happen to my ulanzis

Blueforcer, 2023-11-18:
Can you check this one?
[firmware.zip](https://github.com/Blueforcer/awtrix-light/files/13401532/firmware.zip)

luebbe, 2023-11-18:
The glitches are still there.

Blueforcer, 2023-11-18:
🤷

Can you give me all your JSONs you sent to awtrix?

luebbe, 2023-11-18:
On my way:

```
dev.json:
{
  "min_battery" : 600,
  "temp_dec_places" : 1,
  "temp_offset" : -11.0
}

mqtt:
custom/outsidetemperature
{
  "icon": "2355a",
  "text": "8.1°C"
}

mqtt:
custom/powertotal
{
  "icon": "40828",
  "text": "758W"
}
```

Blueforcer, 2023-11-18:
Mhh no glitches for me... need to dig deeper i guess at some point
[image]

nerdtechse, 2023-11-19:
I tried the beta 0.91B2 and some icons works al the time and some  isnt showing up or shows for 1 sec and the disapears. Icon 21256 from lametric is showing for 1 sec then disappears for example.  
[image]

luebbe, 2023-11-19:
I tried different transition effects. It doesn't play a role. So I think we can rule this out. The icons for custom applications (dis)appear.

Blueforcer, 2023-11-19:
please try current main, according to some discord testers it works now

luebbe, 2023-11-20:
Just upgraded to current main and it looks good. Thank you!

luebbe, 2023-12-02:
For me it is fixed. I think we can close this issue.