Blueforcer/awtrix3 #410: [FEATURE REQUEST] - Notifications in Cyrillic

cimburlen, 2023-11-19:
Hello. Is it possible to add the ability to display notifications in Cyrillic in the next firmware. This makes it very difficult to use with Home Assistant. As far as I understand, most of the fonts here support the Cyrillic alphabet.

Blueforcer, 2023-11-19:
The Awtrix Font doesn't support Cyrillic character.
This would requires to create a new font.
I don't even know if Cyrillic characters would fit in 5 pixel height.

From my side I have no plans to add it. But maybe another user wich has knowledge about that can do this.
Sorry

closed 2023-11-28 by Blueforcer

10der, 2023-12-03:
> The Awtrix Font doesn't support Cyrillic character. This would requires to create a new font. I don't even know if Cyrillic characters would fit in 5 pixel height.
> 
> From my side I have no plans to add it. Sorry. But maybe another user wich has knowledge about that can do this.

[image]

NB:
red right upper it's not artefact, it's mark for red-alert :(