Blueforcer/awtrix3 #411: [BUG] - Unable to hide temperature / humidity and battery charge levels

pacmac, 2023-11-21:
I am unable to use the humidity and temperature display apps as even with adjustment of the offset in dev.json, due to heating of the device itself, the values displayed are wildly inaccurate, and I'm not interested in seeing the battery charge level.

So this means that probably 50% of the time is being used to display data that I don't want to see.

https://github.com/Blueforcer/awtrix-light/blob/e77e875bad1ef7adae63952354f39c1829cc801f/docs/api.md#addremove-and-rearange-apps

That shows that a mqtt publish to deviceid/apps or a api call to api/apps with an array of apps will show or hide the apps, it worked in 0.7 however it appears to have no effect in 0.9 even though the device responds with `OK`

```
[
{"name":"time","show":true},
{"name":"date","show":false},
{"name":"temp","show":false},
{"name":"hum","show":false},
{"name":"bat","show":false},
{"name":"eyes","show":false}
]
```

Can this functionality be restored as I would like to choose what info is being displayed on my system.

pacmac, 2023-11-21:
I also see that the internal app names were changed, I have changed them, the device responds with OK, but the apps are not hidden:

```
[
{"name":"time","show":true},
{"name":"date","show":false},
{"name":"temperature","show":false},
{"name":"humidity","show":false},
{"name":"battery","show":false},
{"name":"eyes","show":false}
]
```

Blueforcer, 2023-11-21:
Please use the current main documentation. There is a reason this API is not public
You have 3 options to disable these apps:
- [Use the on-screen menu with the buttons](https://blueforcer.github.io/awtrix-light/#/onscreen)
- [Use the Andorid or iOS app](https://github.com/Blueforcer/awtrix-light#new-awtrix-app) 
- [Use the Settings API ](https://blueforcer.github.io/awtrix-light/#/api?id=change-settings)

pacmac, 2023-11-21:
Ahh, thanks, I did not notice these:

[image]

closed 2023-11-21 by pacmac