Blueforcer/awtrix3 #413: If al apps have died, last still displayed

salabim, 2023-11-21:
I have a number of apps that all have duration. When all apps have died (reached their duration), the last remains visible. I would like to end with an empty display (essentially meaning that the server doesn't work anymore).

Blueforcer, 2023-11-21:
i dont think **duration** is what you expect..
**duration** just tells awtrix after X seconds it should swich to the next one.

salabim, 2023-11-21:
Sorry, I meant lifetime.

salabim, 2023-11-22:
Let me try and explain better:

I have only two apps running (no standard apps):
```
POST http://192.168.1.53/api/custom params={'name': 'one'} data=b'{"text": "app one", "duration": 1, "lifetime": 10}'
POST http://192.168.1.53/api/custom params={'name': 'two'} data=b'{"text": "app two", "duration": 1, "lifetime": 12}'
```
At t=10, app one dies correctly.
But at t=12, app two does not die, apparently because it's the last survivor.
When I check the loop, I see that app two is indeed still running.

If there were a third app, app two would have died correctly.

Do I do or interpret anything wrong or is it a bug?

m4rquez, 2023-11-22:
If this should work as an indicator that your server is down, _lifetimeMode = 1_ could be an option for you.
The App is then marked with a red border after its lifetime is expired.

salabim, 2023-11-22:
@m4rquez 
Thanks for the reply.
But that's not the problem. The problem is that when all apps have died (reached lifetime), I still see the last one. And that's what I don't want. My 'indicator' would be a blank screen.

m4rquez, 2023-11-22:
I understood, just thought this could be a "workaround" for you.

Blueforcer, 2023-11-28:
Currently Awtrix is not made for showing none app.. Its designed to show at least one app.
Currently its not planned to rework the scaffold.

closed 2023-11-28 by Blueforcer

salabim, 2023-11-29:
I don't think that observation
*Currently Awtrix is not made for showing none app.*
is correct. When I remove all apps, nothing is shown! Just not when the removal happens because of lifetime expiry.

Blueforcer, 2023-11-29:
> I don't think that observation
> *Currently Awtrix is not made for showing none app.*
> is correct.
> 

You're welcome to think so, but as the developer I think I already know what I've programmed 😜
The fact that it works in some situations is at most a side effect, but  never be explicitly developed for it.