Blueforcer/awtrix3 #414: Set WIFI Credentials via USB [FEATURE REQUEST]

ahk97, 2023-11-23:
I would like to use an TC001/Awtrix in a new location.
Id would be nice if its possible to change the WIFI Credentials via the USB Port (serial terminal).
Currently I have to do a full reset of the unit to get it into ap mode again.

Thanks!

Blueforcer, 2023-11-23:
Why full reset?
You can set the new credentials in the Webinterface at any time.
You can also turn off the current wifi, and awtrix will open the AP mode after reboot.

ahk97, 2023-11-23:
Ok. I did not know that.
Then it would probably be sufficient to add a note about the WIFI setup features to the docs.
Or: 6 short button presses start the Wifi Manager / AP Mode (like the tasmota devices)

closed 2023-11-28 by Blueforcer