Blueforcer/awtrix3 #415: [BUG] Lower left pixel flashes yellow constantly

bruschhaupt, 2023-11-23:
# Bug report

## Describe the bug

The left lower pixel flashes yellow continuously. I can't overwrite the pixel either. Does anyone know why this is and how I can switch it off? 

WATCH
[image]

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.90

Blueforcer, 2023-11-23:
That means you configured a mqtt broker but awtrix can't connect.
Remove the broker host in order to disable it.

bruschhaupt, 2023-11-23:
Thank you for the quick solution

closed 2023-11-23 by bruschhaupt