Blueforcer/awtrix3 #416: [BUG] Awtrix Light w/ Ulanzi TC001 w/ Battery Removed Shuts-off After Some Time

misterjtc, 2023-11-25:
# Bug report

## Describe the bug

I have removed the battery from the my Ulanzi TC001 and awtrix appears to be working well but after an hour or two the clock shuts off and I have to unplug it and plug it back in to get it to work again. I have also noticed that the light on the back in initially is red and then when the clock turns off the light is green. I am wondering if it has something to do with voltage regulation with the battery removed.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.9

## To Reproduce

Steps to reproduce the behavior:

1. Remove the battery
2.  Plug in the pixel clock and push the two outer buttons to boot
3. After an hour or two the clock turns off

## Expected behavior

I expect the clock to not shut off

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-11-25:
This has nothing to do with the firmware. Awtrix is unable to turn the hardware on or off. Make sure your power supply is strong enough.

florianru1984, 2023-11-27:
I have the same Issue, since 0.9.0. But i have the battery installed.

Cerunnos75, 2023-11-27:
Same here. But only on 1 of my 2 clocks.
Battery is still installed, tried different power supplies (2.1A; 3A). After 1-2 hrs the clock shuts down :-(

Cerunnos75, 2023-11-28:
UPDATE: I figured out, that (in my case) it was neither the device, nor the power supply. **It was the original cable**.
After testing 5 different power supplies, I wanted to try a re-flash and noticed a permant loss of USB-connection. Using another cable, the connection stayed stable. Using the new cable for power, the clock stayed online now for 24hrs. I will give a new update after some days.

closed 2023-11-28 by Blueforcer